cta-observatory/ctapipe #2216: Problem with path trait inheritance

Tobychev, 2023-01-13:
**Describe the bug**
When constructing larger class hierarchies, child classes do not properly inherit traits defined on the base class (largely eliminating the use of the class hierarchies).

**To Reproduce**
The attached code has been extracted from my current prototype for a benchmarking tool and demonstrates the problem.

```
from ctapipe.core import Tool, traits, Component

class Benchmark(Component):
    """
    Produce intensity benchmark plots
    """

    input_file = traits.Path(
        help="Location and name of input file",
        allow_none=False,
        exists=True,
        directory_ok=False,
        file_ok=True,
    ).tag(config=True)

class DL1aIntensityBenchmark(Benchmark):
    """
    Produce intensity benchmark plots
    """
    camera_type = traits.Unicode(help="the type of camera").tag(config=True)

    def __init__(self, parent=None, config=None, **kwargs):
        super().__init__(parent=parent, config=config, **kwargs)
        self.plots = []
        self.plots.append(PixelNoisePlot(parent=self))

    def make_benchmarks(self):
        for plot in self.plots:
            plot.run(stage="dl1a")

class BenchmarkTool(Tool):
    """
    Create benchmark plots and tables for different stages of the cta-pipe
    workflow.

    The resulting files are saved to a user specified root-directory
    following a fixed hierachy
    """

    name = "ctapipe-benchmark"
    description = traits.Unicode(__doc__)

    classes = [DL1aIntensityBenchmark]

    aliases = {
        ("i", "input"): "Benchmark.input_file",
    }

    def start(self):
        self.plot_maker.make_benchmarks()

    def setup(self):
        self.plot_maker = DL1aIntensityBenchmark(parent=self,)

class BenchmarkPlot(Benchmark):
    """
    Produce intensity benchmark plots
    """

    def run(self, stage):
        self.input_file
        
class PixelNoisePlot(BenchmarkPlot):
    def prepare_data(self):
        pass

    def layout_figures(self):
        pass

tool =BenchmarkTool()
tool.run(argv=["--input","../DATA/gamma_public_merged.dl2.h5"])
```

**Expected behavior**
That the class `PixelNoisePlot` should have `input_file` set'

kosack, 2023-01-13:
I attempted to reproduce this with the following example, but it works fine (both with argv, and passing a Config).  I don't quite see the difference between my example and the one above:
```py3
from ctapipe.core import Tool, Component, traits, run_tool

class Benchmark(Component):
    """Run a benchmark"""

    input_file = traits.Unicode().tag(config=True)
    overwrite = traits.Bool().tag(config=True)

class DL1Benchmark(Benchmark):
    """Benchamrk DL1 things"""

    n_bins = traits.Integer(10, min=1, max=100).tag(config=True)

    def run(self):
        print(self.input_file)

class DL1BenchmarkInternal(DL1Benchmark):
    pass

class DL2Benchmark(DL1Benchmark):
    """Benchamrk DL1 things"""

    n_tels = traits.Integer(10, min=1, max=100).tag(config=True)

    def __init__(self, subarray, config=None, parent=None, **kwargs):
        """example where some other value needs to be passed, like subarray"""
        super().__init__(parent=parent, config=config, **kwargs)
        self.subarray = subarray
        self.dl1 = DL1BenchmarkInternal(parent=self)

    def run(self):
        print(self.input_file)

class AnotherBenchmark(Benchmark):
    pass

class MyTool(Tool):
    classes = []
    aliases = {"input": "Benchmark.input_file"}

    def setup(self):
        self.benchmarks = []
        self.benchmarks.append(DL1Benchmark(parent=self))
        self.benchmarks.append(DL2Benchmark(subarray=None, parent=self))
        self.benchmarks.append(AnotherBenchmark(parent=self))

    def start(self):
        for b in self.benchmarks:
            print(b.get_current_config())
            if hasattr(b, "run"):
                b.run()

    def finish(self):
        print("I'm done")

from traitlets.config import Config

config = Config()
config.Benchmark.input_file = "some_file.h5"
config.DL1Benchmark.input_file = "X"

tool = MyTool(config=config)
tool.run(argv=[])
```
Output is as expected, with no exception thrown (other than SystemExit):
```
{'DL1Benchmark': {'input_file': 'X', 'n_bins': 10, 'overwrite': False}}
X
{'DL2Benchmark': {'input_file': 'X', 'n_bins': 10, 'n_tels': 10, 'overwrite': False, 'DL1BenchmarkInternal': {'input_file': 'X', 'n_bins': 10, 'overwrite': False}}}
X
{'AnotherBenchmark': {'input_file': 'some_file.h5', 'overwrite': False}}
I'm done
```

maxnoe, 2023-01-13:
I think the problem is that something validates the default before the cli arguments are passed and then it fails, can you try allowing None and default_value=None ?

kosack, 2023-01-13:
Tracing through this a bit, the error happens at an unexpected place: it is not an error in constructing the classes, but occurs in `BenchmarkPlot.run()` on the line that says `self.input_file`

I tried changing this to:
```python
   def run(self, stage):
        print(self.__class__.__name__, "RUN", stage)
        print(self.input_file)
        print("END RUN")
```

And get the errror:

```
023-01-13 16:00:46,645 ERROR [__main__.ctapipe-benchmark] (tool.run): The 'input_file' trait of a PixelNoisePlot instance expected a pathlib.Path or non-empty str for a directory or file, not the Sentinel traitlets.Undefined.
2023-01-13 16:00:46,646 ERROR [__main__.ctapipe-benchmark] (tool.run): Use --help for more info
HERE
DONE
PixelNoisePlot RUN dl1a
```

Note that END RUN is never printed, so the exception is arising when the trait `self.input_path` is first accessed.    

Running it in a debugger, I can reproduce this:
```
HERE
DONE
PixelNoisePlot RUN dl1a
> /var/folders/dv/rg9cnf0d3qg6n5khk1d7yjyxqg1x5k/T/ipykernel_38386/4227567625.py(10)run()
      8 
      9         from IPython.core.debugger import Pdb; Pdb().set_trace()
---> 10         print(type(self.input_file))
     11         print(self.input_file)
     12         print("END RUN")

ipdb>  self
<__main__.PixelNoisePlot object at 0x1148a5c70>

ipdb>  self.input_file
*** traitlets.traitlets.TraitError: The 'input_file' trait of a PixelNoisePlot instance expected a pathlib.Path or non-empty str for a directory or file, not the Sentinel traitlets.Undefined.

ipdb>  self.parent
<__main__.DL1aIntensityBenchmark object at 0x1148a5a90>

ipdb>  self.parent.input_file
PosixPath('/Users/kkosack/data.h5')
```
And note that the parent's trait can be accessed, and looks ok.  So somehow the configuration is not bubbling down to this class, but in my example above, i have even deeper a hierarchy and it works.

Tobychev, 2023-01-13:
@kosack I've managed to reproduce the problem with the `input_file` not being set properly in your example too, it took some steps: 
* first by adding a constructor to `DL1Benchmark` that creates a `DL1BenchmarkInternal`
* As a follow on, making `DL1BenchmarkInternal` a direct subclass of `Benchmark` because otherwise the there was an infinite recursion for some reason
* use `tool.run` and directly specifying the input file using the alias `input`
* Put the `DL1Benchmark` into the `classes` list

There might be something more I've forgotten, but here's a modification to your example that will throw the same error:
```
from ctapipe.core import Tool, Component, traits, run_tool

class Benchmark(Component):
    """Run a benchmark"""

    input_file = traits.Path(help="set this").tag(config=True)
    overwrite = traits.Bool().tag(config=True)

class DL1Benchmark(Benchmark):
    """Benchamrk DL1 things"""

    n_bins = traits.Integer(10, min=1, max=100).tag(config=True)
    
    def __init__(self, config=None, parent=None, **kwargs):
        super().__init__(parent=parent, config=config, **kwargs)        
        
        self.dl1 = DL1BenchmarkInternal(parent=self)
    
    def run(self):
        print(f"In {self.__class__}",self.input_file)
        self.dl1.run()

        
class DL1BenchmarkInternal(Benchmark):
    pass

class MyTool(Tool):
    classes = [DL1Benchmark]
    aliases = {"input": "Benchmark.input_file"}

    def setup(self):
        self.benchmarks = []
        self.benchmarks.append(DL1Benchmark(parent=self))

    def start(self):
        for b in self.benchmarks:
            print(b.get_current_config())
            if hasattr(b, "run"):
                b.run()

    def finish(self):
        print("I'm done")

tool = MyTool()
tool.run(argv=['--input','some_file.h5'])
```

kosack, 2023-01-19:
> I think the problem is that something validates the default before the cli arguments are passed and then it fails, can you try allowing None and default_value=None ?

This has no effect - I can reproduce the error using @Tobychev 's example above.  I even removed one level of class inheritance and it still occurs, which is also strange.

maxnoe, 2023-01-19:
When I do `print(self.config)` in `BenchmarkTool.setup`, the config shows up as:
```
{'DL1aIntensityBenchmark': {'input_file': 'dataset://gamma_prod5.simtel.zst'}}
```

Which explains why the other tools do not use the config, it's specific to `DL1aIntensityBenchmark`. 

Adding `Benchmark` to the `classes` of the tool results in 

```
{'Benchmark': {'input_file': 'dataset://gamma_prod5.simtel.zst'}}
```

And this in turn makes the tool work, since now the subclasses are allowed to take that config value.

maxnoe, 2023-01-19:
This is something we should discuss, but the setup above looks a bit strange to me. 

You want that a single command line option propagates into multiple Component's traitlets (running multiple benchmarks on the same file).

However, then I'd think it's much more natural to have the `Tool` load the data and give `Table` to the Benchmark components? Instead of each component again reading a file?

maxnoe, 2023-01-19:
So the error here is not adding all classes that need to be configured into the `classes` of the `Tool` implementation.

maxnoe, 2023-01-20:
We also have the `classes_with_traits` method to automatically include all subclasses, so you should change to:

```
class BenchmarkTool(Tool):
    ...
    classes = classes_with_traits(Benchmark)
```
which also works.

closed 2023-01-24 by Tobychev