cta-observatory/ctapipe #2218: CI Tests failing with Exit Code 134

RuneDominik, 2023-01-13:
**Describe the bug**
Recently, some CI tests seemingly random fail with exit code 134 (probably killed via SIGABRT). Locally all tests complete without the problem.

**To Reproduce**
Run the CI e.g. on #2168 or #2138

**Expected behavior**
Tests should complete.

**Supporting information**
[image]

[image]

**Additional context**
According to [this](https://github.com/actions/runner-images/issues/6709) issue this might mean that the CI exceeds the runners resources (e.g. 7GB RAM). Profiling the memory usage locally using [`mprof`](https://pypi.org/project/memory-profiler/) and simply running `pytest` results in a somewhat monotonic rise in usage reaching 7GB. It might be that the current masters memory usage is pushing the runners capacities and newer PRs simply reach the limit.

#### On current master:
[image]

#### On #2168:
[image]

Quickly comparing this to `gammapy`'s without having the datasets downloaded shows that their memory is freed more often then `ctapipe`'s:

#### On current gammapy master:
[image]

kosack, 2023-01-13:
this certainly looks like a memory problem - we quite recently ran memprof and it showed stable memory use around a few GB (still somewhat higher than gammapy due to the data sizes, but nothing like what you see here). So I suspect there is a regression.

maxnoe, 2023-01-14:
These are the object counts after the last test has run:

```
.dict                       336613
list                       179640
function                   81750
tuple                      77713
weakref                    64178
builtin_function_or_method 49083
Float64Col                 42293
Quantity                   35714
QuantityColumnTransform    33407
cell                       20568
Path                       20130
CompositeUnit              17768
Int64Col                   13299
Float32Col                 10718
Angle                      10384
getset_descriptor          9867
Table                      9711
set                        9536
Int32Col                   8972
type                       8497
wrapper_descriptor         7602
Parameter                  7180
Description                6704
Column                     6693
TelescopeDescription       6019
CameraDescription          6019
MetaIsDescription          5958
method_descriptor          5682
Int16Col                   5631
method                     5360
CameraGeometry             5145
```

It seems somehow most of the subarray data is not cleaned up and also table readers (see all the column transforms) are sticking around

maxnoe, 2023-01-16:
It's actuall 143, which for docker means the container got killed due to resource constraints. So it very certainly is a memory issue.

commit 8a3610f3 (2023-01-18): Use weakref.proxy(parent) in Component.__init__.

Fixes #2218, where the CI fails with OOM, by allowing python
to garbage-collect Tools after tests.
commit 71c7e0fb (2023-01-18): Use weakref.proxy(parent) in Component.__init__.

Fixes #2218, where the CI fails with OOM, by allowing python
to garbage-collect Tools after tests.
commit f24bb1ef (2023-01-18): Use weakref.proxy(parent) in Component.__init__.

Fixes #2218, where the CI fails with OOM, by allowing python
to garbage-collect Tools after tests.
commit 19b3c869 (2023-01-18): Use weakref.proxy(parent) in Component.__init__ and Reconstructor.read

Fixes #2218, where the CI fails with OOM, by allowing python
to garbage-collect Tools after tests.
closed 2023-01-19 by maxnoe