cta-observatory/ctapipe #2226: Faster Docs CI ?

nbiederbeck, 2023-01-19:
Shall we try to reduce the CI time for documentation-only changes?  I think of running just the docs CI when having the label `documentation` attached to the issue, and ignoring the whole pytest suite.

That would mean that we need to split our current pytest approach, where we test "normal" tests and documentation tests simultaneously.

This would greatly reduce the time to passing tests in minor (or large) documentation changes, and we would not be held up by increased memory usage, suddenly incompatible dependencies etc. like we were in the last few days.

kosack, 2023-01-19:
The only problem is that sometimes normal changes have unexpected affects on the docs, since the docs include executed notebooks.  If a change to the code breaks the docs, the PR should not be merged

kosack, 2023-01-19:
Yes that would be fine - changes to the docs only should not require pytest to be run.

PR #2227 (2023-10-31): run test only if labels don't contain 'documentation'
Fix #2226
closed 2023-10-31 by kosack