cta-observatory/ctapipe #2230: Switch to PDGID numbering scheme for particles

maxnoe, 2023-01-20:
We should move from the sim_telarray specific particle numbering scheme to one well-defined and widely used.

The PDGID numbering scheme will be used by CORSIKA 8 and is also proposed in the draft of the common data models document for CTA.

kosack, 2023-01-20:
This should also be synced with [gammasim-tools](https://github.com/gammasim/gammasim-tools) if they don't already do this (I have not checked, but I posted a message on Slack)

maxnoe, 2023-01-20:
What kind of sync do you expect to be necessary? I would expect that the simtel output will still contain simtel numbers but that we convert those in the `SimTelEventSource`.

kosack, 2023-01-20:
I mean they also have a mapping from an internal representation to the corsika numbering, so we consider using the same so we don't end up with configuration mismatches, or confusing monitoring data. 

They responded they only use CORSIKA7 numbering, and I notice they have a mapping to [particle name as well](https://github.com/gammasim/gammasim-tools/blob/a2e5bc0a4acbddc60d1b38f7560e44ce6017c98e/data/parameters/corsika_parameters.yml#L40-L61)

orelgueta, 2023-01-20:
Yes we only use name which are translated to CORSIKA 7 numbers. I guess we will have to keep both numbering schemes in gammasim-tools and translate as necessary when running the simulation. However, as Max said, at the output you will probably have to do the translation as well.

kosack, 2023-01-20:
Would be nice to have a shared set of functions that does the mappings to/from various conventions

maxnoe, 2023-01-20:
https://github.com/scikit-hep/particle/

Also supports CORSIKA 7 (was added recently by a Dortmund master student)

orelgueta, 2024-08-28:
@GernotMaier implemented a small class to accept either PDG ID or CORSIKA 7 ID in simtools. You might find it useful (or at least it would be nice to use the same class if possible).

https://github.com/gammasim/simtools/blob/main/simtools/corsika/primary_particle.py