cta-observatory/ctapipe #2236: Predict on subarrays that differ from training?

nbiederbeck, 2023-01-26:
Do we want to support predicting (applying models) to files with subarrays that differ from the subarrays of the files used in the training of the models? 

The biggest reasons I see:
Yes: just "removing" telesopes shouldn't make a big difference
No: training on Paranal and applying on La Palma

PR with proposed change to **not support**: #2235.

kosack, 2023-01-26:
Sorry I replied in the PR not here (didnt' notice this issue).   I think it is sufficient to:
* check that the list of TelescopeDescriptions is the same
* check that the arrays are in the same location

For the first, it's easy (see comment in the PR), but for the second we dont' yet have a nice way to identify which site is being simulated. So you could either:
1. just check the subarray name and assume that is enough (but that could also change if a sub-subarray is selected)
2. maybe check the EarthLocation  (`subarray.reference_location`)  of the subarray and compare that? (perhaps within some small tolerance, in case it's re-measured)

nbiederbeck, 2023-01-26:
[Karl](https://github.com/cta-observatory/ctapipe/pull/2235#discussion_r1087931533):
Do you really need to check that they are identical subarrays, or rather check that they contain the same telescope types? Since the models are per-telescope-type, the energy reconstruction as is will still work if you use a model trained for 2 LSTs and you give it an array of 4, for example .

And for real observed data, there will likely be cases where we have for example a subarray with a single telescope missing where you still want to reconstruct energy using the original model (for large arrays the difference would be small, even in the case of some sort of more complex stereo model)

So maybe:

    if set(self.loader.subarray.telescope_types) != set(reconstructor.subarray.telescope_types):

[Max](https://github.com/cta-observatory/ctapipe/pull/2235#discussion_r1087940748):

> Do you really need to check that they are identical subarrays, or rather check that they contain the same telescope types?

I'd say it depends on whether stereoscopic features (e.g. HillasReconstructor) have been used or not.

kosack, 2023-01-26:
> I'd say it depends on whether stereoscopic features (e.g. HillasReconstructor) have been used or not.

That just means you rely on a shower geometry, but it should still work if the array changes - you still get a geometry

maxnoe, 2023-01-26:
But you learn on different data than you predict on, which might introduces biases.

But this is something I think we shouldn't enforce at the ctapipe level... this is more about best practices than an actual error. E.g. you might actually want to take that bias instead of having to re-train because one out of 50 telescopes broke.

kosack, 2023-01-26:
Yes, we will  always have biases though, and in real data we have to accept some, so can't be so rigid.  It simply increases th systematics, and may even be calibratable after the fact.   I mean, if we have 70 telescopes, and allow 2 to be off-line, that would be 2415 different IRFs we would have to produce to properly cover it.   

It will be an interesting question for tailored sims: do we re-train models for each run? Do we have enough stats? If so, we could even cover that case

kosack, 2023-01-26:
Maybe this would work:
* if the subarrays are different at all, emit a warning
* if the list of TelescopeDescriptions or the reference_position (if not None) are different, raise an exception