cta-observatory/ctapipe #2237: Force users to think about warnings

kosack, 2023-01-26:
Right now, warnings happen and are logged, but can be lost in a stream of other info if the log level is high. 

Using [warnings.catch_warnings(record=True)](https://docs.python.org/3/library/warnings.html#warnings.catch_warnings), we could print a summary at the end of each Tool's run() function, at minimum:

```
WARNING: 12 warnings were emitted. Please check the logs.
```

At maximum, we could even re-list the warnings.

See e.g. #2236 where a warning may be useful, but shouldn't always be ignored.

Tobychev, 2023-02-03:
Having suffered under the yoke of the HAP warnings, I think having a summary of all warnings like `hap-check` provides is very useful, but it is also important that the user gets some sort of guidance on what to do with the list. 

The ideal world would be that each warning comes with an explanation of why it is raised, an suggestions for further actions. 

Maybe even a system for specifying analysis specific assumptions that would lead to warnings being silenced as the end up expected for a particular set of assumptions, but now I'm just dreaming.

kosack, 2023-02-10:
> The ideal world would be that each warning comes with an explanation of why it is raised, an suggestions for further actions.

That is similar to what we say in the Software Programming Standards document for exceptions. It should apply to warnings as well:

>The exception description shall be written assuming the reader is not an expert in the software. Note that in the case of a problem, exception names and descriptions are what will be seen by the software end-user or operator, who are not experts in the software’s internal design. Therefore, the message shall help them to resolve the problem or to provide a useful bug report