cta-observatory/ctapipe #2248: FeatureGenerator modifies and returns input table

maxnoe, 2023-02-03:
Currently, the `FeatureGenerator` modifies and returns the input table.

This is a counter-intuitive API.
We should either:

* have it mutate the input and not return
* Create a new table and return that

Part of why I thought we need #2240 was that the reconstructors modify the input table, which should probably not be the case.

A couple of solutions I came up with:

* functional approach: return new table with added columns
* mutating approach: remove the return, allow the override

Both have significant drawbacks: I think it is never a good idea to mutate input in that way (reconstructors.predict_table should produce a prediction, not have side effects on the input table). But creating a full copy will also use much more memory.

I think a good solution would be to return a table only containing the new features and havng the `table_to_X` method of the sklearn reconstructures collect features from both the input table and the generated features table to keep both cleanly separated.

kosack, 2023-02-03:
Also pandas, which many ML people use, does not mutate tables in-place - it always returns a new one, so following that convention is a good idea.

e.g. :

```python
df = pd.DataFrame(dict(x=[1,2,3], y=[6,7,8]))

print(df.eval('z=x*y'))

   x  y   z
0  1  6   6
1  2  7  14
2  3  8  24

print(df)  # still the original
   x  y
0  1  6
1  2  7
2  3  8
```  
Is there a large memory overhead of returning a new table? (i.e. is a full copy done inside, or only references to the old columns?)

kosack, 2023-02-03:
> I think a good solution would be to return a table only containing the new features and havng the table_to_X method of the sklearn reconstructures collect features from both the input table and the generated features table to keep both cleanly separated.

This also sounds fine, but wouldn't appending two Tables later use the same memory copying as just returning the full one? Or is it done more efficiently in astropy.table?

Tobychev, 2023-02-03:
> Also pandas, which many ML people use, does not mutate tables in-place - it always returns a new one, so following that convention is a good idea.

Strictly speaking pandas allows for both, because there is almost always a `inplace` option defaulting to false in most of its functions.

maxnoe, 2023-02-03:
> This also sounds fine, but wouldn't appending two Tables later use the same memory copying as just returning the full one? Or is it done more efficiently in astropy.table?

I have to check that. I think it might be possible to share the underlying data for the existing columns, which would remediate that part of the issue.

maxnoe, 2023-02-06:
Checking with astropy, there are two solutions viable here. 

a) Creating a full copy of the table with `table.copy()`, this will of course have basically everything in memory twice.
b) Creating a "shallow" copy of the table with `Table({col: other[col] for col in other.colnames}, copy=False)`, this will create a new table but will not copy the underlying arrays of the existing columns. This is basically what we want, but has the surprising side-effect that the existing columns will be sharing the data: modifying column-data will affect both tables while adding / removing columns will only affect the table which is operated on.

Given the trade-offs and that we don't actually modify column data here, I'd go for b)

PR #2249 (2023-02-09): Refactor apply tool to use HDF5Merger, read data only once
Refactor ctapipe-apply models
    
* The tool now uses the HDF5Merger component for copying data from
the input file, which makes it possible to select what data should
be copied.

* Data is only read once in chunks from the input files and all
reconstructors are applied in order to each of the chunks.

For this, it was needed to not modify the input table in the feature generator, which I opted to do by the shallow copy approach as discussed here: #2248 

Fixes #2248 
Fixes #2130
Fixes #2250
closed 2023-02-09 by maxnoe