cta-observatory/ctapipe #2257: Is clearing 'info' member in CameraGeometry.to_table still needed?

HealthyPear, 2023-02-14:
I am using part of this code to implement SWGO TankReadout geometry and I stumbled upon this artisanal fix,

https://github.com/cta-observatory/ctapipe/blob/dc22a33ea45fd258b552fe158be160663c02dfa8/ctapipe/instrument/camera/geometry.py#L604 

It seems that the [original issue](https://github.com/astropy/astropy/issues/11066) in astropy was fixed sometime ago, so I thought maybe this is not needed anymore.

maxnoe, 2023-02-14:
See https://github.com/cta-observatory/ctapipe/pull/2181

`__getitem__` is still slower without clearing it.

HealthyPear, 2023-02-14:
I see OK. We can close this issue then.

HealthyPear, 2023-02-14:
Or if we want to be picky we might think to update that comment line to the open issues in case someone else finds it

closed 2023-10-21 by maxnoe