cta-observatory/ctapipe #2260: CTA ACTIVITY END TIME doesn't appear in Reference metadata

kosack, 2023-02-16:
**Describe the bug**

the `CTA ACTIVITY STOP TIME` header is missing in output files.  This is because the provenance info that records it happens after DataWriter is already finished, so we never implemented it in DataWriter.  We should maybe just hack it to be written as the time when DataWriter is finished, not getting it from the provenance module, since that is only available after `Provenance().end_activity()` is called. 

**Additional context**

This is not a critical bug, as this info is recorded in the provenance log, but it can be useful info to attach to the file for bookeeping purposes, i.e. to check how long it took to create a specific file without reading the provenance log.  However, the file's last access time could also work there.

maxnoe, 2023-02-16:
> We should maybe just hack it to be written as the time when DataWriter is finished

Are you sure this is still the case in the current master? Because with #1926 , `DataWriter.finish` is the very last thing being called in e.g. `ctapipe-process`, after the `finish_activity()` is called in `Tool.start()`.

maxnoe, 2023-02-16:
Ah, yes, it's missing... but in principle it should not be possible to include it.

```
In [4]: f.root._v_attrs
Out[4]: 
/._v_attrs (AttributeSet), 34 attributes:
   [CLASS := 'GROUP',
    CTA ACTIVITY ID := 'fd06a843-c6ac-453e-89c5-40bb10bd2518',
    CTA ACTIVITY NAME := '/home/mnoethe/.local/conda/envs/cta-dev/bin/python3.9',
    CTA ACTIVITY SOFTWARE NAME := 'ctapipe',
    CTA ACTIVITY SOFTWARE VERSION := '0.18.1.dev6+gde317fb9',
    CTA ACTIVITY START TIME := '2023-02-16 14:17:19.156',
In [4]: f.root._v_attrs
Out[4]: 
/._v_attrs (AttributeSet), 34 attributes:
   [CLASS := 'GROUP',
    CTA ACTIVITY ID := 'fd06a843-c6ac-453e-89c5-40bb10bd2518',
    CTA ACTIVITY NAME := '/home/mnoethe/.local/conda/envs/cta-dev/bin/python3.9',
    CTA ACTIVITY SOFTWARE NAME := 'ctapipe',
    CTA ACTIVITY SOFTWARE VERSION := '0.18.1.dev6+gde317fb9',
    CTA ACTIVITY START TIME := '2023-02-16 14:17:19.156',
```

maxnoe, 2023-02-16:
It's worse... 

Since now we write the provenance after "finish_activity" (like it probably should be), what is written is a blank provenance of a fresh activitity, since `current_acitivity` is None and then a new one is created.

PR #2261 (2023-03-16): Fix wrong activity meta being written into output files
This also adds the stop time.

Fixes #2260
closed 2023-03-16 by maxnoe