cta-observatory/ctapipe #2262: Need to pin numpydoc version

Tobychev, 2023-02-17:
**Describe the bug**
Trying to build docs with versions from `environment.yml` fails with following error
```
Exception occurred:
  File "/local/home/tb273079/miniconda3/envs/cta-dev/lib/python3.9/site-packages/numpydoc/numpydoc.py", line 34, in <module>
    raise RuntimeError("Sphinx 4.2 or newer is required")
```

**To Reproduce**
Steps to reproduce the behavior:
1. make doc

**Expected behavior**
docs should build

kosack, 2023-02-20:
Any idea why this is not causing a test failure?

maxnoe, 2023-02-20:
I guess @Tobychev installed / update single packages locally and in the CI starting from a fresh env we always get it with a correct dependency resolution

closed 2024-07-15 by maxnoe