cta-observatory/ctapipe #2270: Alternative image parametrization based on gaussian fitting

clara-escanuela, 2023-03-01:
The goal of this implementation would be to improve the angular resolution by introducing a image-fitting algorithm. An alternative image parametrization method is presented. 

Using a chi-squared minimization, we want to fit a 2D function to the image with the Hillas parameters as seeds. The function could be a 2D Gaussian, a 2D skewed Gaussian (to model the shower tail) or a skewed Gaussian in the longitudinal direction and a Cauchy function in the transverse direction. 

Image-fitting algorithms have been proven to be successful before (https://arxiv.org/pdf/1211.0254.pdf). Nevertheless, I am unsure on how to proceed with the implementation. Some advice is welcome.

clara-escanuela, 2023-03-01:
I would say that as a first step, I could create a new py file in ctapipe.image which takes the hillas parameters and do the fit.

maxnoe, 2023-03-01:
Hi @clara-escanuela,

I think this is a valuable method to have in ctapipe itself, if only to evaluate how it might improve analysis of truncated showers etc.

You can take a look at how e.g. the `MuonIntensityFitter` is implemented. 

I think the best solution would be to:

* Use the likelihood functions (not chisquare) defined in `ctapipe.image.pixel_likelihood` https://ctapipe.readthedocs.io/en/latest/ctapipe_api/image/pixel_likelihood.html
* Use the models defined in https://ctapipe.readthedocs.io/en/latest/ctapipe_api/image/toymodel.html, if we start using them for fits, we should maybe rename from "toymodel", but it might also be appropriate for this, since it actually is a dramatically simpliefied model of a shower image
* Use iminuit to fit the parameters of the toymodel

This might need some improvements to the toymodels, e.g. a method that actually directly corresponds to a  iminuit loss function. 

I'd add a new `Component` that has the API `__call__(self, event: ArrayEventContainer)` 
that gets the hillas parameters from the event, maybe with a `QualityQuery`.

That could e.g. be used to only run this on showers with non-zero leakage to increase performance.

kosack, 2023-03-01:
Also , you might check with the group in Marseille (CPPM), as I think they implemented exactly this already (they showed sensitivity curves produced with protopipe at multiple ASWG meetings, so the implementation must be ctapipe-compatible). Maybe the code was not in a clean enough state to commit it though?

clara-escanuela, 2023-03-01:
Who, in that group, should I talk to? Could you give me a name?

HealthyPear, 2023-03-01:
@clara-escanuela the person you want to talk with is @gaia-verna .