cta-observatory/ctapipe #2272: Rename confusing Hillas angular parameters

kosack, 2023-03-01:
Right now we have `hillas.phi` and `hillas.psi`, which are somewhat easy to confuse without reading the descriptions. It might be better to rename them to wordier, but more descriptive versions:

Something along the lines of:

Image axis angle:
* `hillas.psi` → `hillas.image_axis_angle`  (or `position_angle` - see below)

Polar coordinates of the centroid:
* `hillas.phi`→ `hillas.centroid_polar_angle`
* `hillas.r` → `hillas.centroid_polar_radius`

"Cartesian" (locally) position of the centroid
*  `hillas.fov_lat` → `hillas.centroid_fov_lat`
* `hillas.fov_lon` → `hillas.centroid_fov_lon`

We might also consider the term  [Position Angle](https://en.wikipedia.org/wiki/Position_angle), which is  how astronomers refer to 2D rotation angles on the sky sphere (not 3D angular coordinates). 

Any other suggestions?

maxnoe, 2023-03-01:
This is a duplicate of https://github.com/cta-observatory/ctapipe/issues/1078

kosack, 2023-03-01:
It's not really a duplicate: I saw that issue, but it relates to the algorithm implementation not the naming of the data model . That one should probably be closed  as the  algorithm was already implemented. The only thing else in that issue is to add more documentation, but it's related.  This issue is more about the fact that we have had a few bugs due to how easy it is to misspell or confuse phi/psi

maxnoe, 2023-03-01:
> It's not really a duplicate: I saw that issue, but it relates to the algorithm implementation not the naming of the data model .

Strange, the first sentence in that issue is:

> In a recent PR it was suggested that the Hillas parameters are renamed to be more self-descriptive (https://github.com/cta-observatory/ctapipe/pull/771). I think this was a good suggestion, so I am creating this issue so that it is not forgotten.

kosack, 2023-03-01:
Yeah, the title of that issue is confusing,  It references however [this comment](https://github.com/cta-observatory/ctapipe/pull/771#issuecomment-409503867), which is basically this issue. We can maybe close #771 and #1078 (though 1078 also has some interesting info in it, even if not a naming scheme).

This issue was meant to replace both,

maxnoe, 2023-10-21:
Still think that #1078 is exactly the same as this and it has the relevant discussions, so I am closing here

closed 2023-10-21 by maxnoe