cta-observatory/ctapipe #2321: StereoTrigger does only work when tels_with_trigger is a python list, real sources use numpy arrays

maxnoe, 2023-04-27:
**Describe the bug**

The code assumed and the unit tests made it so that the `StereoTrigger` works only when `tels_with_trigger` is a python list,
however the event source implementations provide a `np.ndarray` with dtype `uint16`.

**To Reproduce**
Try processing a prod3 or prod6 divergent pointing file where there might be single LSTs with the stereo trigger set to simulate the LST hardware stereo trigger.

**Expected behavior**

Single LSTs are removed, events with less than 2 telescopes are completely discarded.

**Supporting information**

**Additional context**

This is needed for the prod6 divergent processing, since due to an error in the simtel configuration, the LST hardware stereo trigger was not actually already included in the simulation.

kosack, 2023-05-04:
Can you explain how single LST events got through here? Looking at the code, if a numpy array was given, it looks like i should simply raise an exception since np.array.remove() doesn't exist.  Maybe I'm missing something. Or is it just that we never had it find an event that needed to be skipped since the stereo trigger was properly simualted?

maxnoe, 2023-05-05:
Exactly. In prod5b, the stereo trigger is simulated.  Due to a bug in the config,  it is missing in the prod6 divergent pointing simulations.

PR #2320 (2023-05-10): Fix SoftwareTrigger for actual simtel files
Fixes #2321
closed 2023-05-10 by maxnoe