cta-observatory/ctapipe #2326: Minor change to tailcuts cleaning  to improve the detection of signal pixels

clara-escanuela, 2023-05-10:
The tailcut cleaner could be modified in order to add pixels above the (low) boundary threshold if they continuously connect to a pixel above the (high) picture threshold, rather than only the first neighbour. This would require a very small change in the code. It should be equivalent to the hysteresis thresholding implemented in scikit-image (https://scikit-image.org/docs/dev/api/skimage.filters.html#skimage.filters.apply_hysteresis_threshold, reference https://ieeexplore.ieee.org/document/4767851) .

[image]
[image]
[image]

maxnoe, 2023-05-10:
That's a nice idea but I'd say not a *small* change to tailcuts.

THat rather seems like it's own cleaning method as an adaption of the normal tailcuts.

I'd be in favor of adding it though (or as an option to the existing tailcuts)

clara-escanuela, 2023-05-11:
I thought it could be added either as an option in tailcuts or a new cleaning method (although it is very similar to tailcuts)