cta-observatory/ctapipe #2335: Changes to simulation containers

maxnoe, 2023-05-17:
**Please describe the use case that requires this feature.**

In a discussion with @kbernloehr in the gitlab, it came up that the `injection_height` in the `mcrunheader` object, which we store in the `SimulationConfigContainer` is not used by simtel array and exists only for historical reasons, it is always -1.

The real injection height, is stored for each shower in the `MCShower` object, but currently we don't store this.

**Describe the solution you'd like**

Remove `injection_height` from the `SimulationConfigContainer` and add `injection_height` to the `SimulatedShowerContainer`.

kbernloehr, 2023-05-17:
No, please not again anything with 'height' as that still implies the wrong units.

kosack, 2023-05-22:
good point @kbernloehr : let's at least pick a standard name for atmospheric depth units. What is the best term for a unit compatible with g/cm^2?  Is it "_depth_" (also confusing as can be in distance units), "_slant depth_", "_grammage_", just "_X_", "_column density_"?

kbernloehr, 2023-05-22:
The CORSIKA User's Guide calls it 'starting grammage' and 'mass overburden' (of the starting altitude). The terms 'slant depth' and 'column density' seem to imply something along the direction of the primary particle. Same for 'X0', given that 'Xmax' is counted along the incoming primaries path. 'Grammage' may be ambiguous as well and the fact that Wikipedia knows it as a term from the paper industry does not help much either. 'Overburden' sounds about right, making it clearer that it counts what is (vertically) above. (For Wikipedia it is mainly a term from the mining industry but it also mentions that 'the overburden of an underground laboratory may be important to shield the facility from cosmic radiation'. Oh well, those underground guys count it in meters of water equivalent but at least it is in the right direction.)

maxnoe, 2023-06-02:
I'd choose `starting_grammage` then, it's the section title of this quantity in the CORSIKA user guide and I think the most common word for that in our community.

PR #2343 (2023-07-18): Remove SimulationConfigContainer.injection_height, add SimulatedShowercontainer.starting_grammage
Fixes #2335
closed 2023-07-18 by maxnoe