cta-observatory/ctapipe #2344: true image rarely missing for bright events, results in ctapipe-process error

maxnoe, 2023-06-05:
**Please describe the use case that requires this feature.**

It seems that due to a bug or a memory limitation, sometimes the true image data block is completely missing in sim-telarray files. If writing the true images, that results in an exception as the DataWriter tries to write a `None` value into the table setup for writing the true images.

This was reported by @TjarkMiener, observed on 4-LST SiPM and PMT camera simulations, but it seems also observed by Matthieu Heller in the SST-1M processing.

**Describe the solution you'd like**

We should maybe fill these missing true images with -1 or zero values instead of None to allow processing of such files.

A more general solution would be to *optionally* catch exceptions in the event-loop and continue processing, however it is maybe hard to guarantee a well-defined state of the output files then, especially if like here the error happens during a write operation and some values are already written to the file for the failing event.

kosack, 2023-06-05:
Probably should explicitly define an exception for this rather than catching all in the event loop. It's better to expect things are correct and fix issues as they come, rather than to make it very robust to failing, but potentially have some unexpected results.   And a warning should be issued to the log that the event was missing the true_image, when it was expected. 

I guess this has been reported to Konrad so it can be fixed in simtelarray?  As usual, I'd prefer to avoid too may software fixes for upstream bugs, as it makes the code more complex and becomes difficult to maintain and potentially error prone (lesson learned from past IACT software).  It's the same for bad telescope data: it's not our problem to fix it only report it.   But we could have a temporary (to be deprecated) fix until the bug is solved and the sims are re-run.

maxnoe, 2023-06-05:
> I guess this has been reported to Konrad so it can be fixed in simtelarray?

Yes, we reported it, but not sure if this is really a bug... It seems to be that if something is too large for the memory limitations, it is not written and the simulation is continued.

But yes, Konrad is investigating further I think. I will also ask Luisa if this issue appears in any official CTA productions.

> Probably should explicitly define an exception for this rather than catching all in the event loop. It's better to expect things are correct and fix issues as they come, rather than to make it very [...]

Catching this specific exception here is I think not enough, since this would leave the file in a bad state (partially written event data).

I think if we decide that this should be handled in ctapipe, then the correct thing to do would be to fill something different than None (i.e. `np.full(n_pixels, -1)` in case the true images are missing for only some events.

Actually, we used to fill all-zeros for missing true_image, but that was changed some tmie ago.

kbernloehr, 2024-11-26:
There are two conditions where a the photo-electron list may be missing from the data file:

1. Weak images. Normally, the separate threshold to record the true p.e. list for a telescope should be low enough that the probability of this happening is negligible. It could still happen that NSB pushes a weak image over the trigger threshold.
2. Extremely bright images. Rather than exiting sim_telarray when the p.e. list does not fit into the pre-allocated location, or re-allocating that array and risking that the total memory exceeds the job limits, skipping the p.e. list is an attempt to recover from/after such a problematic event. Note that the MAX_PHOTOELECTRONS (camera-wide) and MAX_PIXEL_PHOTOELECTRONS (per pixel) environment variables are available to set appropriate limits. Where such a problem happens frequently, you better adapt these limits and make sure the job has sufficient resources allocated.

In the first case, the resulting simulated and triggered event is perfectly OK but in the second case, the image will be based on only those p.e. until the limit was reached. Since any further p.e. cannot be kept in the assigned memory location and, therefore, cannot be sorted into pixels, the simulated image does not include them. If that only happens when pixels are highly saturated, the image might still be usable - but it better should not be trusted. The limit might be reached from photons emitted in the first half of the shower, with the tail of the shower remaining dark in the image.

As this is handled telescope by telescope, such an overflow might affect just one telescope or a few telescopes and data for the other telescopes is OK. Skipping the entire event would not be the proper solution. Recording data for the overflow telescope(s) like nothing happened, just that the apparent number of p.e. happens to match MAX_PHOTOELECTRONS, would be dangerous. Thus, in any case you need to catch overflow telescopes and exclude them from the analysis - perhaps keeping the fact that there was an overflow in mind. While the missing, clipped, p.e. list may be an indicator of the problem - why record a huge list of photo-electrons when knowing that it is incomplete and unusable? - the real indicator should be that the number of photo-electrons is equal to the allowed maximum. The log file, for sure, will have that event mentioned.

maxnoe, 2024-11-27:
Thanks for your detailed response @kbernloehr, as always.

This is the structure of one of the files in question in Prod6:

`/vo.cta.in2p3.fr/MC/PROD6/LaPalma/gamma/sim_telarray/4749/Data/000xxx/gamma_20deg_0deg_run000065___cta-prod6-2156m-LaPalma-dark+magic.simtel.zst`

```
 MCShower[2020](shower_id=16641)
 MCEvent[2021](event_id=1664100)
 MCEvent[2021](event_id=1664101)
 TelescopeData[1204](event_id=1664101)
     PhotoElectrons[1208](array_id=1, telescope_id=1)
 MCEvent[2021](event_id=1664102)
 MCEvent[2021](event_id=1664103)
 MCEvent[2021](event_id=1664104)
 MCEvent[2021](event_id=1664105)
 MCEvent[2021](event_id=1664106)
 TelescopeData[1204](event_id=1664106)
     PhotoElectrons[1208](array_id=6, telescope_id=0)
     PhotoElectrons[1208](array_id=6, telescope_id=1)
     PhotoElectrons[1208](array_id=6, telescope_id=2)
     PhotoElectrons[1208](array_id=6, telescope_id=3)
 MCPhotoelectronSum[2026](event_id=1664106)
 ArrayEvent[2010](event_id=1664106)
     TriggerInformation[2009](event_id=1664106)
     TelescopeEvent[2201](telescope_id=1)
         TelescopeEventHeader[2011](telescope_id=1)
         ADCSamples[2013](telescope_id=1)
         ImageParameters[2014](telescope_id=1)
         PixelList[2027](telescope_id=1, code=1, kind=selected)
     TelescopeEvent[2202](telescope_id=2)
         TelescopeEventHeader[2011](telescope_id=2)
         ADCSamples[2013](telescope_id=2)
         ImageParameters[2014](telescope_id=2)
         PixelList[2027](telescope_id=2, code=1, kind=selected)
     TelescopeEvent[2232](telescope_id=32)
         TelescopeEventHeader[2011](telescope_id=32)
         ADCSamples[2013](telescope_id=32)
         PixelTiming[2016](telescope_id=32)
         PixelList[2027](telescope_id=32, code=0, kind=triggered)
     TrackingPosition[2101](telescope_id=1)
     TrackingPosition[2102](telescope_id=2)
     TrackingPosition[2103](telescope_id=3)
     TrackingPosition[2104](telescope_id=4)
     TrackingPosition[2105](telescope_id=5)
     And 26 objects more of the same type
     StereoReconstruction[2015](result_bits=000000000101)
 MCEvent[2021](event_id=1664107)
``` 

From what you wrote above, this seems to be the first case? A dim image?

maxnoe, 2024-11-27:
The true_image_sum is 18 for this event:
```
In [32]: print(e.simulation.tel[32])
{'impact': {'distance': <Quantity 173.49648623 m>,
            'distance_uncert': <Quantity 0. m>},
 'true_image': array([-1, -1, -1, ..., -1, -1, -1], dtype=int32),
 'true_image_sum': np.int32(18),
 'true_parameters': None}```

This is the reconstructed image:

![magic_img_no_true_image](https://github.com/user-attachments/assets/0bf57592-17ad-4efa-b782-fb59cf3308ad)

kbernloehr, 2024-11-27:
Even though such events could have been avoided by a little difference in configuration (see below), ctapipe should better be prepared to handle them. This particular image will certainly not pass any reasonable event selection. It could give you another point in the reconstructed p.e. versus true p.e. curve, at the very low end, but I don't think you can learn much from it. Therefore, I am not bothered at all by it existence.

Before explaining why this, with the current prod-6 setting, could only happen for a MAGIC telescope, let me briefly review my recommended strategy for setting up related parameter MIN_PHOTONS, MIN_PHOTOELECTRONS, and STORE_PHOTOELECTRONS. For the first two you have corresponding histograms allocated and filled in sim_telarray, before applying any trigger, after applying telescope trigger, and after applying the array trigger on top of that.

You could start with single-telescope events, mono trigger, or with a small array, like four telescopes, and stereo trigger.
The first processing step is to compare the sum of photon bunch sizes, a floating-point number representing the number of photons hitting the telescope fiducial sphere, against MIN_PHOTONS and ignore the telescope if is below the threshold. Photon ray-tracing and detection only if this first threshold is met. The resulting number of detected photo-electrons needs to equal or larger than MIN_PHOTOELECTRONS before simulating any trigger or readout signal traces.
Therefore, the first test production starts with MIN_PHOTONS=0; MIN_PHOTOELECTRONS=0 and we look at the lowest numbers resulting in any triggered events. Since the 'real' production can be expected to have way more events, we better apply some (downward) margin then. Say, no event was seen to trigger with less than 1000 photons and no trigger with less than 50 p.e. With MIN_PHOTONS=400 and MIN_PE=25 we can expect to hardly ever miss an event that would trigger while not wasting too much CPU time on those that have absolutely no chance of triggering.

The strategy for STORE_PHOTOELECTRONS should simply be to be less or equal to MIN_PHOTOELECTRONS. With less p.e. the signals are not simulated and there can be no trigger. But a tool like simtrig could re-simulate the camera response and the trigger, based on the p.e. list and the NSB p.e. rates. If you want to be prepared for that and think that a more advanced trigger may have a 10% lower threshold, set STORE_PHOTOELECTRONS a bit below MIN_PHOTOELECTRONS. The only telescope type where the prod-6 settings did not follow that recommendation, was MAGIC:

```
Telescope   min_photons    min_pe(*)    store_pe(**)
---------   -----------    ---------    ------------
LST         300            25           20
MST-FC      360            25           22
MST-NC      450            30           26
SST         250            12           10
SCT         600            25           20

MAGIC1      100            15           20
MAGIC2      100            15           20

----
(*)  Parameter name: MIN_PHOTOELECTRONS
(**) Parameter name: STORE_PHOTOELECTRONS
```
In the example event, with 18 p.e., we are going into the signal and trigger simulation(>=15 p.e.) - and it just happened to trigger by chance - but skipped storing the p.e. list (<20 p.e.). If we had used STORE_PHOTOELECTRONS=12 for the MAGICs, rather than 20, we would have been on the safe side, storing its p.e. list, but had missed that case for which ctapipe needs to be prepared. (Totally unintended, I promise. But sooner or later it would have happened, so better prepare for it now.) Setting all of these parameters to zero, you would have been totally on the safe side - but spent perhaps a factor ten or fifty more CPU time in the telescope simulation, without gaining a single triggered event. And recorded tons and tons of p.e. lists for such non-triggered events.

[By the way, I noticed two other oddities in this comparison, which deserve to be looked into. But I think the downward safety margins applied were large enough and I am not worried.]

maxnoe, 2025-04-02:
The fix in #2659 only partially adressed this issue.

For large productions with options as mentioned above, it can occur reasonably often, that the first event is such an event. In this case, the `DataWriter` does not setup the `true_image` column (as it is missing from the first event) and thus no true images are written at all for this input file.

This creates an issue later when merging files, as some will have true images and some not.

PR #2659 (2024-12-04): Fix error in ctapipe-process in case telescope event is missing true image
Fixes #2344
commit 7e7119f1 (2025-04-16): Improve handling of missing true images, fixes #2344
closed 2025-04-16 by kosack