cta-observatory/ctapipe #2360: CameraGeometry gives a single value for geom.pixel_width, but CameraDisplay expects an array.

mdpunch, 2023-06-20:
**Describe the bug**
When creating a new CameraGeometry from scratch, from an array of pix_x, pix_y, where these are evenly spaced pixels, the geom.pixel_width which is filled automatically is a single value.
However, the CameraDisplay seems to expect an array, and gives an index error
```
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
Cell In[291], line 1
----> 1 CameraDisplay(geom_new)

File /scr/punch/miniconda3/envs/nectar-dev/lib/python3.8/site-packages/ctapipe/visualization/mpl_camera.py:125, in CameraDisplay.__init__(self, geometry, image, ax, title, norm, cmap, allow_pick, autoupdate, autoscale, show_frame)
    123 pix_x = self.geom.pix_x.value[self.mask]
    124 pix_y = self.geom.pix_y.value[self.mask]
--> 125 pix_width = self.geom.pixel_width.value[self.mask]
    127 for x, y, w in zip(pix_x, pix_y, pix_width):
    128     if self.geom.pix_type == PixelShape.HEXAGON:

IndexError: invalid index to scalar variable.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run the attached code...

**Expected behavior**
Expect the code to apply the `geom.pixel_width` to all pixels, if it is not an array, instead of giving an error.

We can change the single value to an array "by hand" as in the final comment in the attached code, but this shouldn't be needed, I think.

[make_simple_geoms.ipynb.zip](https://github.com/cta-observatory/ctapipe/files/11800453/make_simple_geoms.ipynb.zip)

mdpunch, 2023-06-20:
Indeed, I give a single value for pixel_area.  But that name is singular!  So I expected to give one value...

maxnoe, 2023-06-20:
id, x, y, are also singular...

The docstring is pretty clear:
```
    pix_area : array(float)
        surface area of each pixel, if None will be calculated
```

I'll add validation logic to provide a good area immediately

mdpunch, 2023-06-20:
Although the docstring says:
```
pix_area: array(float)
    surface area of each pixel, if None will be calculated
```
... so there, an array is indeed expected.

mdpunch, 2023-06-20:
Although, although, the docstring is incorrect, in that it doesn't calculate the `pix_area` if I give no value:
```
TypeError: __init__() missing 1 required positional argument: 'pix_area'
```