cta-observatory/ctapipe #2367: Check table length consistency in ctapipe-merge tool

TjarkMiener, 2023-06-23:
**Describe the bug**
(I think) the _ctapipe-merge_ tool lets me merge broken files with different table lengths.
 
**To Reproduce**
Steps to reproduce the behavior:
1. I was processing thousands of runs to DL1b via the _ctapipe-process_ tool. Slurm failed during the processing so some of the tables are not completely written, i.e. broken files but with reasonable file sizes. I resumed the processing for the missing runs and the runs with low file sizes. However, there were still broken DL1b files. 
2. The _ctapipe-merge_ tool with the option _skipbrokenfile_ lets me merge all of the files without a problem.
3. When applying the model the tool failed because of different table lengths at the end of processing.

**Expected behavior**
Either break or skip in _ctapipe-merge_ tool when a broken file with different table length is present.

kosack, 2023-06-23:
There was similar issue that came up in the merging of the divergent pointing sims.  

*  how to determine that a run is "broken"?:  right now this is written to the provenance log, but we don't record any header information in the h5 file with the activity status. That could be done, and then it would be easy to check if a file was "completed" vs "cancelled" or "failed" (or missing the metadata altogether).  For now, a hacky way would be to check for the existence of the `/simulation/service/shower_distribution` table, which is only written if the file is complete and no "max-events" was requested.  

*  We discussed making a *ctapipe-filecheck* tool that looks for common data-model inconsistencies and mis-matches, like repeated obs_ids, etc.  That would not really be the same thing as what we need to avoid merging bad runs, but would help to identify problems.

kosack, 2023-06-23:
@TjarkMiener it might be helpful if you can reproduce this with a minimal set, e.g. 2 runs merged the exhibit this behavior, and then provide the broken files so we can test?

TjarkMiener, 2023-07-06:
@kosack @maxnoe Sorry for the delay! I put two broken files (run nr. 525 and 537) and one completed file (run nr. 1) in the lst-cluster `/fefs/aswg/workspace/tjark.miener/issue2367/`. I'm working with `ctapipe v0.19.3.dev1+g2c142cf`. You can reproduce the behavior by:
```
 $ ctapipe-apply-models --input=/fefs/aswg/workspace/tjark.miener/issue2367/electron_run1.dl1.h5 --reconstructor stereoPMT/energy.pkl --output testing.h5 --overwrite 
// worked
$ ctapipe-apply-models --input=/fefs/aswg/workspace/tjark.miener/issue2367/electron_run525.dl1.h5 --reconstructor stereoPMT/energy.pkl --output testing.h5 --overwrite
// error: [ctapipe.ctapipe-apply-models] (tool.run): Caught unexpected exception: 'HillasReconstructor_tel_impact_distance'
$ ctapipe-apply-models --input=/fefs/aswg/workspace/tjark.miener/issue2367/electron_run537.dl1.h5 --reconstructor stereoPMT/energy.pkl --output testing.h5 --overwrite
// error: [ctapipe.ctapipe-apply-models] (tool.run): Caught unexpected exception: 'HillasReconstructor_tel_impact_distance'
$ ctapipe-merge -i /fefs/aswg/workspace/tjark.miener/issue2367/ -p "electron*" -o merged_corruptedfiles.h5 --progress --overwrite
// worked
$ ctapipe-apply-models --input merged_corruptedfiles.h5 --reconstructor stereoPMT/energy.pkl --output testing.h5 --overwrite
//error: [ctapipe.ctapipe-apply-models] (tool.run): Caught unexpected exception: Tables must have identical length
```