cta-observatory/ctapipe #2372: Remove ctapipe-dump-triggers

maxnoe, 2023-07-04:
This just writes timestamps of events into a table that is not compatible with the rest of ctapipe. 

If there is a need for such a tool, we should adapt the `ctapipe-process` tool to just write out the trigger information in the datawriter without doing any actual processing, but I doubt anyone really needs this.

kosack, 2023-07-04:
I think we get this info in any case in a ctapipe output file, but an option to only write it would work as well and it would be in the standard format so ctapipe-merge, etc would work.    @mdpunch are you still using this tool?

mdpunch, 2023-07-04:
> I think we get this info in any case in a ctapipe output file, but an option to only write it would work as well and it would be in the standard format so ctapipe-merge, etc would work. @mdpunch are you still using this tool?

Hi Karl @kosack ,

I haven't used this in a while, as I have the file I produced with it, and just use that.

If there is a way to get this from the DL0 data (or R1?), that would be fine.  But this relates to our parallel discussion, that it would be good to get the trigger-times also for the "busy" telescopes, and if possible for mono-telescope triggers or a fraction thereof (for my study, I just "reinjected" random mono triggers times until I reached the expected telescope rate, which is not the most ideal).

In short, if there is a way to access the "SWAT data" after the change request which is now accepted: 
https://gitlab.cta-observatory.org/cta-computing/computing-change-request-management/-/issues/3#note_123771 
then there should be no further need for this tool, I think?

mdpunch, 2023-07-04:
(for @maxnoe if you're curious, I use this as MC input to the Trigger Array S(t)imulator (TATS) which produces trigger pulses for various "Telescopes" with the correct DeltaTs, which then get time-stamped by our bench of UCTS-TiCkS boards, which send their time-stamp streams to the SWAT provided at the time by Jerzy Borkowski... as the first real-time test of the SWAT on MC data... see my talk at the Bologna in 2018 https://indico.cta-observatory.org/event/2249/contributions/21603/ )

maxnoe, 2023-07-05:
I think we are mixing several things here, especially the content of simtel output and the planned ACADA DL0. Sure, these should be as similar as possible, but I think simtel cannot simulate dead time of telescopes, at least not in any meaningful manner, since it processes one shower at a time and you would have to specify spectra of cosmic rays and the source when simulating.
So I think at least for the foreseeable future, `tels_with_data` will always be identical to `tels_with_trigger` for simulations.

For the observed DL0, the information you want will be written by ACADA into a single file per observation block, giving trigger times for all telescopes, also the busy ones, and the difference between `tels_with_trigger` and `tels_with_data` will tell you which telescopes were busy. At least that's how I understood this change request and the file defintions in the _draft_ of the ACADA-DPPS ICD.

So for now, for your use case of getting relative trigger times between the telescopes of one air shower, I suggest we add options to `ctapipe-process` so it can only store the simulation parameters and the trigger information without doing any actual processing and remove the `dump-triggers` tool.

Should be very simple.

moralejo, 2023-07-05:
> I think we are mixing several things here, especially the content of simtel output and the planned ACADA DL0. Sure, these should be as similar as possible, but I think simtel cannot simulate dead [...]

I think we should have simtel always triggering on the shower (btw we recently noticed that the LST in simulation sometimes triggers early on afterpulses, and an otherwise observable shower is lost because it falls outside of the readout window).
The per-telescope dead time can be introduced at the analysis level by just randomly dropping some images, at a rate which will depend on the dead time & rates of the real data one intends to analyze. I see no advantage in introducing that dead time already at the simulation level.

kosack, 2023-07-05:
> I think we should have simtel always triggering on the shower (btw we recently noticed that the LST in simulation sometimes triggers early on afterpulses, and an otherwise observable shower is lost because it falls outside of the readout window).

Isn't that what you would want, as the same will happen in real data? Or alternately couldn't you simply not run the simulations with after-pulsing enabled?

maxnoe, 2023-07-05:
> Isn't that what you would want, as the same will happen in real data? 

I think this is an issue because the simulation can only result in a single event per air shower. In reality, you can trigger on noise and then after the dead time, you can trigger again on the shower, creating two events.

The current simulations seem to be hard wired to result in a single event and then the effect is very dependent on how long the actually simulated time range is. The longer the time range, the higher the probability to trigger on noise rather the shower.

But in analyzing a whole LST simulation production, we found only a handful of such events (@moralejo can probably quantify better), so I expect that this effect is very, very tiny.

mdpunch, 2023-07-05:
I agree with @maxnoe that the simulations (_per se_) shouldn't worry about the dead-time.

The simulations treat one shower event at a time, and it is certainly a bad idea to create correlations between events there.  It would require the events to be strictly random in energy and time, and would probably be messed up by throwing a shower several times at different impacts, or by importance sampling, or whatever.

In my side discussion with Karl @kosack , about how to work on the array dead-time, my suggestion was to have a separate procedure - a "busy injector" - which could take the simulation data, then using an "expected" previous event, to decide if some of the telescopes should be dropped as "expected busy".  This would allow to calculate IRFs with or without expected deadtime.

For the "expected" previous events: The array trigger patterns from the simulations could be used to make a library from which the "busy injector" could take patterns at random - corrected for importance sampling if necessary - to decide if some telescopes should be busy.  The lack of mono-telescope triggers in the simulations is a bit of a barrier to this, especially as Konrad thinks that in the simulations the mono events will be underestimated (but I think this is to be determined).

Anyway, I guess we're straying from the topic of the issue... but it is interesting for me!

Maybe a future topic for the ASWG, as Karl suggested to me?

mdpunch, 2023-07-05:
> The per-telescope dead time can be introduced at the analysis level by just randomly dropping some images, at a rate which will depend on the dead time & rates of the real data one intends to analyze. I see no advantage in introducing that dead time already at the simulation level.

Yes, as I said I don't see the advantage of doing the busy in the simulations, but just randomly dropping telescopes will miss the **correlations** between neighbouring telescopes which are likely to be busy from a previous array event.

mdpunch, 2023-07-05:
> So for now, for your use case of getting relative trigger times between the telescopes of one air shower, I suggest we add options to `ctapipe-process` so it can only store the simulation parameters [...]

Simple is good, so I am in favour of this !

PR #2375 (2023-07-06): Remove dump-triggers, enable ctapipe-process to store only trigger and simulation data
fixes #2372 

Running `ctapipe-process -i <input> -o <output> --no-write-parameters` will result in an output file only containing trigger and simulation information, no any actual processing will happen.
closed 2023-07-06 by kosack