cta-observatory/ctapipe #2380: Remove vitables from environment.yml

StFroese, 2023-07-11:
The environment is currently not installable on M1 macOS since ViTables has no macOS M1 support. I opened an issue there (https://github.com/uvemas/ViTables/issues/115) but unfortunately the repo seems quite inactive. 

It's pretty annoying to comment the according line in the environment.yml out each time I want to install a new environment and remove the comment again when pushing. Does anybody of you have an alternative to vitables in mind which we could use? Otherwise I would suggest to remove it from the environment.yml completely since this is already just the dev environment and each dev can install it on demand.

kosack, 2023-07-11:
It's fine with me.  There are also sometimes missing packages for it conda, so I usually install it in a separate environment to keep it working.  I think it's still developed, but not at a very fast rate. 

However, I would suggest not just removing it, but also adding some info in the documentation about how to install it and use it, since it is very useful for understanding the file structure.

StFroese, 2023-07-12:
Works now for 3.9, I added the osx-arm64 build to the feedstock

closed 2023-07-12 by StFroese