cta-observatory/ctapipe #2382: Allow requiring a minimum angle between crossing lines in hillas reconstructor

maxnoe, 2023-07-14:
**Please describe the use case that requires this feature.**

From a comment by @GernotMaier via email:

EventDisplay uses:
> - min angle between two image lines >20 deg

Since for very small angles, the error on the intersection gets very large, this makes sense and we should think about also including such a condition into ctapipe.

kosack, 2023-07-17:
This would just minimally mean a second _QualityQuery_, but on the intersection list rather than the input images.  Right now, those are computed and stored internally in the HillasReconstructor. 

We could even consider putting them in a Container in the event structure to make them exposed outside the algorithm if that would help.   EventDisplay also uses the scatter in intersection points as an output parameter, so making this explicit could help there as well.

maxnoe, 2023-07-17:
>  EventDisplay also uses the scatter in intersection points as an output parameter, so making this explicit could help there as well.

That is what we put in as error estimate, right?