cta-observatory/ctapipe #2390: x_max to h_max conversion in AtmosphereDensityProfile

kosack, 2023-08-31:
**Please describe the use case that requires this feature.**

Currently we have `atmprof.line_of_sight_integral(h, zenith)` that converts *h* (distance in m) to *x* (slant depth in g cm-2).  For the ImPACT analysis, we need the inverse transform as well. 

**Describe the solution you'd like**

Implement `derivative()` functions for all `AtmosphereDensityProfiles`, similar to the existing `integral()` and provide a top-level function to do the `line_of_sight_derivative()`

maxnoe, 2023-08-31:
Is it really derivative? The inverse function of an integral is not the derivative...

kosack, 2023-08-31:
true, not the best name

maxnoe, 2024-07-15:
This was fixed: https://github.com/cta-observatory/ctapipe/pull/2422

closed 2024-07-15 by maxnoe