cta-observatory/ctapipe #2395: h_max is inconsistent between HillasIntersection and HillasReconstructor

gschwefer, 2023-09-14:
Both HillasReconstructor and HillasIntersection calculate a quantity h_max and return it in the ReconstructedGeometryContainer, but they are inconistent in the physical meaning of h_max: For HillasIntersection, it is the vertical height of the shower maximum above the array, for HillasReconstructor it is the distance of the shower maximum location from the array center, i.e. kind of a "slant height" 

Whatever h_max really should be, it should of course be the same everywhere.

gschwefer, 2023-09-14:
See the related discussion in #2305 about what quantity returned from ImPACT

gschwefer, 2023-09-14:
Also, should the Hillas reconstructors in addition to h_max return the atmospheric slant depth xmax?

kosack, 2023-09-18:
Yes, I believe we should return both `h_max` (which should really be the height of shower max, the z coordinate), and the distance to shower max (along the line of sight).  Slant depth (what we call in the atmosphere module "column density") is then the distance  transformed to a grammage using an atmosphere model, whereas `x_max` is the `h_max` transformed to grammage units.

gschwefer, 2023-09-18:
Ok, I have one question: Do I understand you correctly that you want to give the name `x_max` to the vertical grammage equivalent to `shower_max_height` ?

maxnoe, 2023-09-18:
I think that would be bad, since e.g. Auger uses x max for the depth of the shower maximum.

kosack, 2023-09-18:
We might want to avoid x_max altogther if that is the case. Corsika/SimTelArray define it as a height (mass overburden), not corrected for the zenith angle.  If Auger uses a different definition, it's clear that it is not well defined and we should pick names that are obvious.

Also "depth" as a word usually does *not* contain the cos factor (think water depth in the ocean, which is not a line-of-site measurement), so it's not even clear to me that Auger's definition is different.  Normally, "depth", "height" and "altitude" (when not an angle)  all mean really just the z-coordinate of a point in a fluid/gas defined from different 0 points. 

The path-length from the observer to a point in a medium is different: it is the slant-depth, column-density, etc. It implies an integration over a path, which in our case is a line.

kosack, 2023-09-18:
What we are describing is a point - the shower maximum.   Here is a summary:

- solid red line: integrating along a path that starts at infinity and ends at the shower maximum point is what we currently call the "shower depth" , sometimes called "mass overburden"
- solid black line: in distance units from the ground, this is the altitude or height of shower maximum
- dotted black line: the distance from the observatory to the shower max forms a line, the "line-of-sight", the distance from the ground to the shower max along this line is the "shower distance" (unless you have a better name?) 
- dotted red line: integral along the line-of-sight from infinity to the shower maximum is the _slant depth_ or _column density_ 
 
[image]

maxnoe, 2023-09-18:
>  Corsika/SimTelArray define it as a height (mass overburden), not corrected for the zenith angle.

Actually it depends... CORSIKA has a compile time option for it (SLANT), if set, longittudinal development is sampled in slant depth, if not in vertical depth, which what determines the value of the true xmax, since that comes from a fit to that distribution.

I guess AUGER is of course always using SLANT. We are probably not, but maybe we should? At least for high zenith angles?

kosack, 2023-09-18:
Yes, I mentioned that in the other issue: in CTA, we do not use the SLANT option.

kosack, 2023-09-18:
In the atmosphere module, the density profiles are all defined in terms of height from ground, and each defines an integral along the height direction (so everything is 1D). The `profile.line_of_sight_integral()` does the 2D integration from infinity to a point, along the line-of-sight.   It currently  assumes a cartesian atmosphere for simplicity, but a proper numeric integral could be done for a spherical atmosphere or even a more realistic oblate spheroid if we really wanted, though the difference is small below 60° zenith angle.  That's just an implementation detail, and probably not worth worrying about yet.

maxnoe, 2023-09-19:
Also related: https://github.com/cta-observatory/ctapipe/issues/1030

kosack, 2023-09-20:
Yes, I thought of tat as well. Here's a better diagram of what we have now...:
[image]

kosack, 2023-09-20:
Anyhow, I suggest the following:

For the definition of height, there are two options:
1. define heights as relative to the observatory's center point (0,0,0), e.g. consistent with the current definition of the telescope heights, which are also relative to the array center point (in x,y,z), where a telescope at height 10m is 10m above the array plane.  Then in the conversion to x_max, the suer has to be careful to add the observatory height.
2. add `source.subarray.reference_location.geodetic.height` to `h_max`, so heights are real heights above sea level.  however, we could leave telescope z coordinates relative, which still makes sense  and won't break the reconstruction.

Probably option 2 is safest. 

Then fix the following bugs / add features:
* HillasReconstructor [currently returns h_max](https://github.com/cta-observatory/ctapipe/blob/3c5b4f3516d83d757f92f914da5ab8cd82c39e16/ctapipe/reco/hillas_reconstructor.py#L415C15-L429) as the distance to the shower_max from the _array center_ (0,0,0). This is just wrong.  we should just take the z-coordinate (and add the reference height) and then it is a true "height". If we want the distance in meters to be stored additionally, we could compute the distance from the impact point on the ground projected to sea level, but I'm not sure that is very useful for anything.
* define x_max as just the vertical depth of the shower max (could rename it to `shower_depth` as mentioned above to avoid confusion, at the expense of changing the data model)
* define the slant depth as the line integral from infinity to the shower max point along the line defined by the _shower zenith angle_ (not the pointing one).  Name that `shower_slant_depth` everywhere. 
* make a helper function that uses atmosphere module and the shower reconstruction to compute the real shower depth and slant depth (and vice versa) so we have one place where it is done correctly
* Fix `HillasIntersection` which does a few bad things: it does add the observatory depth (but as a hard-coded number which is wrong!), and it also calls the depth in m "x_max" internally, which is even more confusing. It never actually does the conversion

If that sounds ok, I'll make a preliminary PR to fix at least this in the HillasReconstructor.  So far nothing other than ImPACT actually uses x_max or h_max, so that can be fixed separately.

maxnoe, 2023-09-20:
Sounds good to me

kosack, 2023-09-20:
i did run into the issue that not all of our test files provide a subarray reference_location... So adding the height fails for those. Either we should update the example files, or have a fallback

maxnoe, 2023-09-20:
> i did run into the issue that not all of our test files provide a subarray reference_location... So adding the height fails for those. Either we should update the example files, or have a fallback.

The obs level height should be available for all and could be used as fallback. Specifying the reference location via simtel metadata is fairly new (Prod6 prototype files)

kosack, 2023-09-21:
> We could maybe fill the reference location with dummy lat/lon of 0 but the correct altitude from obslevel in that case:
https://en.wikipedia.org/wiki/Null_Island

That sounds reasonable. I'll open a separate issue or PR for that. For now I fixed the problem by just switching what file is used for generating the dummy subarray in the tests.

kosack, 2023-09-21:
@maxnoe  Where can I access `obslev`? Is it only in `corsika_inputcards`, or is it parsed out somewhere?

PR #2403 (2023-10-23): Fix h_max definition
Fixes #2395 

* In HillasReconstructor, Define h_max as the vertical height *above sea level* (including the observatory height) of the shower-max point, and not the distance from the array which was there before. 
* In HillasIntersection, use the correct observatory height value

When #2402  is merged, I will remove the checks for missing reference_location
closed 2023-10-23 by maxnoe