cta-observatory/ctapipe #2398: Time cleaning with DBSCAN

clara-escanuela, 2023-09-20:
I was not planning to raise this issue yet but looking at the program of the in-person meeting, I have seen that image cleaning is one of the topics. I think we are missing an algorithm that was implemented in HESS by Simon Steinmassl with very good results. I implemented and tested it myself with ctapipe and it seems to work pretty good. We perform a pre-cleaning cut in the image intensity with respect to the noise level in each pixel. The reconstructed time and positions of pixels that pass this cut go through DBSCAN which then identifies high dense regions of points with similar time and space distributions. One of the positive things of this method is that clusters are very well captured, which could be exploited for gh separation. The main difficulty is the parameter optimization. In the plots, I show the survival probability of signal pixels and the distribution of the number of noise pixels retained.
[signal.pdf](https://github.com/cta-observatory/ctapipe/files/12669839/signal.pdf)
[noise.pdf](https://github.com/cta-observatory/ctapipe/files/12669840/noise.pdf)

maxnoe, 2023-09-20:
Sounds good. Since you already implemented and tested it, please feel free to open a pull request.

kosack, 2023-09-20:
Yes, in fact we are missing lots of algorithms for cleaning, in particular those that use the pedestal variances. So this fits well.  I can add a talk in the meeting for you as well if you like

clara-escanuela, 2023-09-20:
Sure. I could give a brief talk about this