cta-observatory/ctapipe #2441: Disp fills table with missing "tel" prefix, results in duplicated columns with and without tel

maxnoe, 2023-10-31:
**Describe the bug**

We made a change to add a `_tel` between the container prefix and the field name to distinguish mono predictions from stereo predictions when loaded into the same table, code here:
https://github.com/cta-observatory/ctapipe/blob/main/ctapipe/reco/utils.py

We do this also with the per-telescope disp output, which has no stereo equivalent. 

The reconstructor is also filling the table without this prefix, so the value with the `tel_` prefix is added.

**To Reproduce**

Run the unit tests for the disp reconstructor, inspect the output:

```
['disp_parameter_norm',
 'disp_parameter_is_valid',
 'disp_parameter_tel_norm',
 'disp_parameter_tel_is_valid']
```

**Expected behavior**

Only have the `disp_paremeter_norm` and `disp_parameter_is_valid` columns

Tobychev, 2023-10-31:
I think this is closed by #2440, right?

maxnoe, 2023-10-31:
Yes, forgot to link it

closed 2023-10-31 by maxnoe