cta-observatory/ctapipe #2515: Is the line_of_sight_integral function really correct?

maxnoe, 2024-02-23:
Another question about the functions currently there: Is the line_of_sight_integral function really correct? Because the distance you put into it is a distance from the observer, but the integral functions uses heights a.s.l.

_Originally posted by @gschwefer in https://github.com/cta-observatory/ctapipe/issues/2422#issuecomment-1790265905_

kosack, 2024-02-23:
yes, the slant depth is what is computed in the line_of_site_integral.  It just uses the chain rule for a flat atmosphere, as shown in the notes.

kosack, 2024-02-23:
Sorry, didn't read the full question correctly.  Yes, we should check that distance to observer is used correctly.  I corrected it in the h_max code, I think, but likely there is something still missing (like the distance from the observer projected along the LOS to sea level)

kosack, 2024-02-23:
It think the best is just to change the help string to be correct: it is not the  distance from the observer, but from sea level.   It's actually the height above sea level of the point you want to turn into a grammage.  Then it's consistent witth how we now define h_max (which is from sea level).   From that height, you integrate out to infinity along the zenith direction.   Calling it "height_to_slant_depth" would probably be more clear as well

gschwefer, 2024-02-23:
Alright I looked at this now...The circle (vertical) overburden ->height a.s.l. ->(vertical) overburden is fully implemented via the functions "integral" and "height_from_overburden". These are what's actually used in all the internal calculations. The functions "height_from_slant_depth" and "line_of_sight_integral" are only convenience functions to account for the zenith angle automatically and not used in any internal calculations. I agree with you, Karl: I will rewrite the 
"line_of_sight_integral" so it's the proper inverse of "height_from_slant_depth" and integrate this in #2422

kosack, 2024-03-04:
I think you should also refactor ImPACT to use the "convenience" functions, since there is an implicit assumption  on the atmosphere shape that is used, and we may improve that in the future. So really, these are not convenience functions: they implement a specific atmosphere model that may change in a future improvement, and thus all Reconstructors should use those functions directly and not implement their own.   For example, we know currently for zenith angles > 70°, this   model is incorrect.  CORSIKA employs a trick for these higher zenith angles, keeping a flat atmosphere, and we could implement that in the future, or even move to a fully spherical atmosphere, but doing so would break ImPACT if it has it's own internal implementation of how to compute slant depth.

gschwefer, 2024-03-04:
Yes, I will of course address these things in ImPACT. So if you want to keep this issue open for that, that's alright, but it should not block the merging of #2422, right?

kosack, 2024-03-05:
> Yes, I will of course address these things in ImPACT. So if you want to keep this issue open for that, that's alright, but it should not block the merging of https://github.com/cta-observatory/ctapipe/pull/2422, right?

No, that can be a separate PR of course! 

By the way, reposting this image from the previous issue/discussion, since it illustrates this issue, where the distance to the maximum should be along the shower-axis from sea level. 
[image]

closed 2024-04-05 by maxnoe