cta-observatory/ctapipe #2564: Problems with ctapipe installation using pip on M1 Mac

jstvdk, 2024-06-14:
**Describe the bug**
Can't install `ctapipe` using `pip` on the M1 Mac. Was tested on several machines. It crashes when trying to install `tables` package. 

It causes problems while installing other packages with `pip`, that depends on `ctapipe`, e.g. I couldn't install `calibpipe` in `pip` editable mode, because it was crushing while installing dependencies of `ctapipe`.

**To Reproduce**
In fresh conda or pyenv environment:

```sh
git clone https://github.com/cta-observatory/ctapipe.git
cd ctapipe
pip install -e .
```

or  more simply from any dir without cloning:
```sh
pip install ctapipe
```
**Supporting information**
```sh
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... done
  Installing backend dependencies ... done
  Preparing editable metadata (pyproject.toml) ... done
Collecting astropy<7.0.0a0,>=5.3 (from ctapipe==0.21.0)
  Using cached astropy-6.1.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (10 kB)
Collecting bokeh~=3.0 (from ctapipe==0.21.0)
  Using cached bokeh-3.4.1-py3-none-any.whl.metadata (12 kB)
Collecting docutils (from ctapipe==0.21.0)
  Using cached docutils-0.21.2-py3-none-any.whl.metadata (2.8 kB)
Collecting eventio<2.0.0a0,>=1.9.1 (from ctapipe==0.21.0)
  Using cached eventio-1.12.1.tar.gz (855 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... done
Collecting iminuit>=2 (from ctapipe==0.21.0)
  Using cached iminuit-2.26.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (11 kB)
Collecting joblib (from ctapipe==0.21.0)
  Using cached joblib-1.4.2-py3-none-any.whl.metadata (5.4 kB)
Collecting matplotlib~=3.0 (from ctapipe==0.21.0)
  Using cached matplotlib-3.9.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (11 kB)
Collecting numba>=0.56 (from ctapipe==0.21.0)
  Using cached numba-0.60.0-cp312-cp312-macosx_11_0_arm64.whl.metadata (2.7 kB)
Collecting numpy~=1.16 (from ctapipe==0.21.0)
  Using cached numpy-1.26.4-cp312-cp312-macosx_11_0_arm64.whl.metadata (61 kB)
Collecting psutil (from ctapipe==0.21.0)
  Using cached psutil-5.9.8-cp38-abi3-macosx_11_0_arm64.whl.metadata (21 kB)
Collecting pyyaml>=5.1 (from ctapipe==0.21.0)
  Using cached PyYAML-6.0.1-cp312-cp312-macosx_11_0_arm64.whl.metadata (2.1 kB)
Collecting requests (from ctapipe==0.21.0)
  Using cached requests-2.32.3-py3-none-any.whl.metadata (4.6 kB)
Collecting scikit-learn!=1.4.0 (from ctapipe==0.21.0)
  Using cached scikit_learn-1.5.0-cp312-cp312-macosx_12_0_arm64.whl.metadata (11 kB)
Collecting scipy~=1.2 (from ctapipe==0.21.0)
  Using cached scipy-1.13.1-cp312-cp312-macosx_12_0_arm64.whl.metadata (60 kB)
Collecting tables~=3.4 (from ctapipe==0.21.0)
  Using cached tables-3.9.2.tar.gz (4.7 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [13 lines of output]
      /var/folders/_p/3fmt5jz5231_9gxtm8xt76p00000gn/T/H5close7r0k17bk.c:1:10: fatal error: 'H5public.h' file not found
      #include "H5public.h"
               ^~~~~~~~~~~~
      1 error generated.
      cpuinfo failed, assuming no CPU features: 'flags'
      * Using Python 3.12.3 | packaged by Anaconda, Inc. | (main, May  6 2024, 14:46:42) [Clang 14.0.6 ]
      * Found cython 3.0.10
      * USE_PKGCONFIG: False
      * Found conda env: ``/Users/vdk/mambaforge/envs/test_env``
      .. ERROR:: Could not find a local HDF5 installation.
         You may need to explicitly state where your local HDF5 headers and
         library can be found by setting the ``HDF5_DIR`` environment
         variable or by using the ``--hdf5`` command-line option.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

**Additional context**
Machine - MacBook Air M1
OS Version - 12, 13, 14.5

maxnoe, 2024-06-14:
This is not an error with ctapipe. The failure happens when you install pytables, a dependency of ctapipe for working with HDF5 files.

tables currently does not provide a pre-built binary wheels macos arm64. 

So it tries to compile from source, which fails because you don't have the necessary build requirements installed.

The easiest solution for you would be to use conda and install the conda-forge builds of pytables.

E.g.

```
$ mamba create -n calibpipe python3=3.12 ctapipe
$ conda activate calibpipe
```

maxnoe, 2024-06-14:
See https://github.com/PyTables/PyTables/issues/1165

closed 2024-06-14 by maxnoe

mexanick, 2024-06-14:
@maxnoe while this is not a `ctapipe` bug, can we propagate the solution (install `ctapipe` with `conda`) into the documentation?

maxnoe, 2024-06-14:
The docs already recommend installing ctapipe via conda-forge

https://ctapipe.readthedocs.io/en/latest/user-guide/getting-started.html#how-to-get-the-latest-version

morcuended, 2024-06-14:
An extra note for Mac users would be good

maxnoe, 2024-06-14:
I was hoping pytables makes a release soon, the build infrastructure is already there.... If the release is not done by next week or so, I'll add something to the docs