cta-observatory/ctapipe #2568: Expired deprectation in `scipy-1.14` results in broken `CameraGeometry`

mexanick, 2024-06-26:
Following the recent release of `scipy`(https://github.com/scipy/scipy/releases/tag/v1.14.0),
the `CameraGeometry` is now broken:
```
self = CameraGeometry(name='LSTCam', pix_type=PixelShape.HEXAGON, npix=1855, cam_rot=-0.000 deg, pix_rot=0.857 rad, frame=<TelescopeFrame Frame (telescope_pointing=None, obstime=None, location=None)>)
    @lazyproperty
    def neighbor_matrix(self):
>       return self.neighbor_matrix_sparse.A
E       AttributeError: 'csr_matrix' object has no attribute 'A'
```
It is an expired deprecation, see:
https://github.com/scipy/scipy/issues/21049

As a result, we need either to promote this issue to a bug (and fix it), or pin the scipy version to be ~1.13

mexanick, 2024-06-26:
I guess, a fix could be simply to use `csr_matrix.toarray()` instead of `csr_matrix.A`
https://docs.scipy.org/doc/scipy-1.12.0/reference/generated/scipy.sparse.csr_matrix.toarray.html

maxnoe, 2024-06-26:
@mexanick Yes, exactly, done here: #2569

commit 29037901 (2024-06-26): Support scipy 1.14, fixes #2568
commit c459e2fc (2024-06-26): Support scipy 1.14, fixes #2568
commit 322a71c7 (2024-06-26): Merge pull request #2569 from cta-observatory/scipy_1.14

Support scipy 1.14, fixes #2568
closed 2024-06-26 by maxnoe