cta-observatory/ctapipe #2574: Use lru_cache instead of custom caching

kosack, 2024-07-03:
https://github.com/cta-observatory/ctapipe/blob/b63a7d4de5d65520cdf721d8b56b8990e43c2673/src/ctapipe/instrument/camera/geometry.py#L934-L935

The hand-crafted border pixel caching could just use the `@functools.lru_cache` or `@functools.cache` decorator. Is there a reason not to?

maxnoe, 2024-07-03:
`lrc_cache`, at least in the past` didn't play nice with class methods.

The cache was global, not per instance.

maxnoe, 2024-07-03:
https://docs.python.org/3/faq/programming.html#faq-cache-method-calls

seems to be supported now

maxnoe, 2024-09-11:
> The two principal tools for caching methods are [functools.cached_property()](https://docs.python.org/3/library/functools.html#functools.cached_property) and [functools.lru_cache()](https://docs.python.org/3/library/functools.html#functools.lru_cache). The former stores results at the instance level and the latter at the class level.

So yes, I'd say this is the reason: we don't want to keep `CameraGeometry` instances alive indefinitely only because their "get_border_pixels" method was called once.

closed 2024-09-11 by maxnoe