cta-observatory/ctapipe #2592: Tests fail with python 3.10 and pip on multiple operating systems

ctoennis, 2024-07-17:
Me and Tjark had tests on our pull requests fail running with python 3.10 on different operating systems. 

The relevant logs are [here ](https://github.com/cta-observatory/ctapipe/actions/runs/9971435793/job/27552561499?pr=2554) for the StatisticsExtractor PR and [here ](https://github.com/cta-observatory/ctapipe/actions/runs/9972275316/job/27555202127?pr=2543) for the VarianceExtractor PR. 

In both cases the issue lies with setuptools not being found (see [here](https://github.com/cta-observatory/ctapipe/actions/runs/9972275316/job/27555202127?pr=2543#:~:text=%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%20ERRORS%20%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D,%3A5%3A%20ModuleNotFoundError))

maxnoe, 2024-07-17:
This has to be a change in pytest test collection logic. For some reason it thinks that `ctapipe/_dev_version/__init__.py` is a test module.

maxnoe, 2024-07-17:
Turns out something must have changed in the `doctest` collection logic, it now also tries to pick up the `_dev_version` file and fails.

Fixed by ignoring the file for doctest.

mexanick, 2024-07-19:
I'm afraid only a part of the problem is solved.
macOS-14, py:3.12 with mamba gives
```
2024-07-19T06:31:24.4422990Z     [0m[96mself[39;49;00m._prov[[33m"[39;49;00m[33msystem[39;49;00m[33m"[39;49;00m].update(_get_system_provenance())[90m[39;49;00m
2024-07-19T06:31:24.4423240Z [1m[31msrc/ctapipe/core/provenance.py[0m:343: in _get_system_provenance
2024-07-19T06:31:24.4423410Z     [0mpackages=_get_python_packages(),[90m[39;49;00m
2024-07-19T06:31:24.4423650Z [1m[31msrc/ctapipe/core/provenance.py[0m:309: in _get_python_packages
2024-07-19T06:31:24.4424290Z     [0m[94mfor[39;49;00m p [95min[39;49;00m [96msorted[39;49;00m(distributions(), key=[94mlambda[39;49;00m d: (d.name [95mor[39;49;00m [33m"[39;49;00m[33m"[39;49;00m).lower())[90m[39;49;00m
2024-07-19T06:31:24.4424650Z [1m[31msrc/ctapipe/core/provenance.py[0m:309: in <lambda>
2024-07-19T06:31:24.4425360Z     [0m[94mfor[39;49;00m p [95min[39;49;00m [96msorted[39;49;00m(distributions(), key=[94mlambda[39;49;00m d: (d.name [95mor[39;49;00m [33m"[39;49;00m[33m"[39;49;00m).lower())[90m[39;49;00m
2024-07-19T06:31:24.4425790Z [1m[31m../../../micromamba/envs/cta-dev/lib/python3.12/importlib/metadata/__init__.py[0m:457: in name
2024-07-19T06:31:24.4426210Z     [0m[94mreturn[39;49;00m [96mself[39;49;00m.metadata[[33m'[39;49;00m[33mName[39;49;00m[33m'[39;49;00m][90m[39;49;00m
2024-07-19T06:31:24.4426450Z _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
2024-07-19T06:31:24.4426460Z 
2024-07-19T06:31:24.4426660Z self = <importlib.metadata._adapters.Message object at 0x111e150d0>
2024-07-19T06:31:24.4426740Z item = 'Name'
2024-07-19T06:31:24.4426750Z 
2024-07-19T06:31:24.4427050Z     [0m[94mdef[39;49;00m [92m__getitem__[39;49;00m([96mself[39;49;00m, item):[90m[39;49;00m
2024-07-19T06:31:24.4427200Z     [90m    [39;49;00m[33m"""[39;49;00m
2024-07-19T06:31:24.4427450Z     [33m    Warn users that a ``KeyError`` can be expected when a[39;49;00m
2024-07-19T06:31:24.4427820Z     [33m    mising key is supplied. Ref python/importlib_metadata#371.[39;49;00m
2024-07-19T06:31:24.4427940Z     [33m    """[39;49;00m[90m[39;49;00m
2024-07-19T06:31:24.4428190Z         res = [96msuper[39;49;00m().[92m__getitem__[39;49;00m(item)[90m[39;49;00m
2024-07-19T06:31:24.4428590Z         [94mif[39;49;00m res [95mis[39;49;00m [94mNone[39;49;00m:[90m[39;49;00m
2024-07-19T06:31:24.4428700Z >           _warn()[90m[39;49;00m
2024-07-19T06:31:24.4429090Z [1m[31mE           DeprecationWarning: Implicit None on return values is deprecated and will raise KeyErrors.[0m
2024-07-19T06:31:24.4429100Z 
2024-07-19T06:31:24.4429500Z [1m[31m../../../micromamba/envs/cta-dev/lib/python3.12/importlib/metadata/_adapters.py[0m:60: DeprecationWarning
```

mexanick, 2024-07-19:
full log for the reference can be found here: https://github.com/cta-observatory/ctapipe/actions/runs/10003501784/job/27650544386?pr=2594

mexanick, 2024-07-19:
I think the problem is that now DeprecationWarning is thrown for every broken package:

https://github.com/python/cpython/blob/e7130dc014c672bfc64782afbf048ffe42363c1f/Lib/importlib/metadata/_adapters.py#L10-L16

and 

https://github.com/python/cpython/blob/e7130dc014c672bfc64782afbf048ffe42363c1f/Lib/importlib/metadata/_adapters.py#L53-L61

We convert DeprecationWarnings to errors, right? @maxnoe

maxnoe, 2024-07-19:
Yes, but we can ignore specific ones that we can do nothing about

maxnoe, 2024-07-19:
Ah, but this one is real and should be fixed.

mexanick, 2024-07-19:
no, we need to be ready that KeyError will be thrown (not yet), it should be taken care in the lambda function you've introduced in #2593

PR #2593 (2024-07-17): Fix traitlets test by ignoring test modules and _dev_version
Fixes #2592
commit efc9090c (2024-07-19): Fix deprecation warning in provenance, fixes #2592
commit fcfef310 (2024-07-19): Fix deprecation warning in provenance, fixes #2592
commit f9108ac6 (2024-07-19): Merge pull request #2599 from cta-observatory/fix_deprecation_warning

Fix deprecation warning in provenance, fixes #2592
closed 2024-07-19 by maxnoe