cta-observatory/ctapipe #2614: Add image extraction parameters to the metadata of the dl1 images table

TjarkMiener, 2024-09-09:
**Please describe the use case that requires this feature.**
For calculating the pedestal level (per sample), the charge median needs to be divided by the window width of the extractor. Since charge extraction and calculation of camera calibration coefficients are now separated, we'd need to propagate the information of the image extraction parameters in the metadata. 

**Describe the solution you'd like**
Add the parameters of the image extractor in the metadata.

**Describe alternatives you've considered**
Use the same config file for the `process`-tool and camcalib-calculation-tool and read the parameters from the config.

**Additional context**
Maybe the information is already stored somewhere in the data format. If so, please point me there. Thanks!

kosack, 2024-09-09:
> Use the same config file for the process-tool and camcalib-calculation-tool and read the parameters from the config.

That sounds error prone since you would have to ensure you sync the config files, so I agree this should be propagated in the table metadata or as a table like `/configuration/process`.   This is also a more general problem: right now the pipeline configuration info only goes into the provenance log, which is not really intended to be read by another processing step, so it would be good to write it somewhere properly.

maxnoe, 2024-09-09:
This is basically the same as #1610 I would say

TjarkMiener, 2024-10-03:
> This is basically the same as #1610 I would say

Yes, I agree! Can you please provide me with a time estimate for #1610 implementation?