cta-observatory/ctapipe #2620: ctapipe-process all calibration fields skipped

BastienLacave, 2024-09-20:
**Describe the bug**
When using ctapipe-process on real LST data,  all calibration, pedestal and flatfields fields are skipped and the calibration therefore crashes. (see .txt attached).

**To Reproduce**
Steps to reproduce the behavior:
1. On run 2929, date 20201118 (see config file attached):

>pedestal : /fefs/aswg/data/real/monitoring/PixelCalibration/Cat-A/drs4_baseline/20201118/v0.9.2/drs4_pedestal.Run02920.0000.h5

>calibration : /fefs/aswg/data/real/monitoring/PixelCalibration/Cat-A/calibration/20201118/v0.9.2/calibration_filters_52.Run02921.0000.h5

>time calibration : /fefs/aswg/data/real/monitoring/PixelCalibration/Cat-A/drs4_time_sampling_from_FF/20191124/v0.9.2/time_calibration.Run01625.0000.h5

>pointing : /fefs/onsite/monitoring/driveLST1/DrivePositioning/DrivePosition_log_20201118.txt

>run summary : /fefs/aswg/data/real/monitoring/RunSummary/RunSummary_20201118.ecsv

>R0 file : /fefs/aswg/data/real/R0/20201118/LST-1.1.Run02929.0000.fits.fz

Command:
ctapipe-process -t 1 --overwrite --input={file} --output={output_file} --config={args.config} --write-parameters --EventTypeFilter.allowed_types=SUBARRAY --progress

This happens on some runs and not others, I have checked that the calib files are not empty and the fields skipped are in the files. I find the calibration files using the onsite package from lstchain.

[ctapipe_standard_pmt_config_02929.json](https://github.com/user-attachments/files/17070776/ctapipe_standard_pmt_config_02929.json)
[sba0.sh.39890718.txt](https://github.com/user-attachments/files/17070791/sba0.sh.39890718.txt)

BastienLacave, 2024-09-20:
I forgot to mention that the same run works but with calib files found in the "pro" directory, but I am interested by the files in the "v0.9.2" directory, is there any reason this works for one and not the other ?

morcuended, 2024-09-20:
> The same run works but with calib files found in the "pro" directory, but I am interested by the files in the "v0.9.2" directory, is there any reason this works for one and not the other ?

Just one note, `pro` directory points to the latest version of calibration produced with lstchain v0.10

maxnoe, 2024-09-20:
You cannot use hdf5 files produced with ctapipe 0.12 (lstchain 0.9) with ctapipe versions newer than 0.16 (lstchain 0.10 uses lstchain 0.19).

This is not a bug, it was a conscious breaking change.

closed 2024-09-20 by maxnoe

BastienLacave, 2024-09-20:
Ah very well then, thank you