cta-observatory/ctapipe #2621: Import error for ctapipe.io after updating to numpy 2.02

GernotMaier, 2024-09-23:
**Describe the bug**
`from ctapipe.io import write_table` throws an ImportError after updating to numpy 2.02

**To Reproduce**
Steps to reproduce the behavior:

```
>>> from ctapipe.io import write_table
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/io/__init__.py", line 17, in <module>
    from .simteleventsource import SimTelEventSource
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/io/simteleventsource.py", line 22, in <module>
    from ..calib.camera.gainselection import GainChannel, GainSelector
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/calib/__init__.py", line 5, in <module>
    from .camera import CameraCalibrator, GainSelector
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/calib/camera/__init__.py", line 6, in <module>
    from .calibrator import CameraCalibrator  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/calib/camera/calibrator.py", line 18, in <module>
    from ctapipe.image.extractor import ImageExtractor
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/image/__init__.py", line 43, in <module>
    from .muon import (
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/image/muon/__init__.py", line 7, in <module>
    from .fitting import kundu_chaudhuri_circle_fit
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/ctapipe/image/muon/fitting.py", line 3, in <module>
    from iminuit import Minuit
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/iminuit/__init__.py", line 24, in <module>
    from iminuit.minuit import Minuit
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/iminuit/minuit.py", line 6, in <module>
    from iminuit import util as mutil
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/iminuit/util.py", line 11, in <module>
    from iminuit import _repr_html, _repr_text, _deprecated
  File "/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/iminuit/_deprecated.py", line 2, in <module>
    from numpy import VisibleDeprecationWarning
ImportError: cannot import name 'VisibleDeprecationWarning' from 'numpy' (/Users/maierg/micromamba/envs/simtools-dev/lib/python3.11/site-packages/numpy/__init__.py)

**Expected behavior**

no error.

**Supporting information**

**Additional context**

maxnoe, 2024-09-23:
It looks like you only updated numpy and not minuit?

There are definitely tests covering the code that is throwing the error here. 

Does this happen in a fresh environment? It might be that old minuit versions were not compatible with numpy 2.0 but didn't specify so in their dependencies.

maxnoe, 2024-09-23:
Side note: we really should make the imports in io lazy. Importing half of ctapipe for write_table is not great.

GernotMaier, 2024-09-23:
This is after running a `micromamba update -f environment.yml` on an existing environment. Let me try to generate a fresh environment to see if somehow minuit was not updated.

maxnoe, 2024-09-23:
Fyi: the first iminuit version to support numpy 2.0 is 2.26 from June:

https://scikit-hep.org/iminuit/changelog.html#june-03-2024

GernotMaier, 2024-09-23:
OK - there is no error when installing a fresh environment. So this is then likely an issue of mamba when updating an env.

I therefore suggest to close this issue.

maxnoe, 2024-09-23:
It's an issue of the dependency specification of the iminuit package, nothing much mamba can do about it.