cta-observatory/ctapipe #2660: Subarray trigger does not match telescopes with data in prod6 files

maxnoe, 2024-11-28:
This breaks some assumptions in `SimTelEventSource` / `DataWriter` / `TableLoader` / `HDF5EventSource` and results in errors
since the length of the trigger table does not match the length of the dl1 / dl2 tables.

Investigating why this happens, I think this might be related to `RANDOM_MONO` feature of sim_telarray active in prod6

maxnoe, 2024-11-28:
In the output file of sim_telarray, the event that causes this mismatch is "weird".

Here is the structure:

```
 MCEvent[2021](event_id=1548602)
 TelescopeData[1204](event_id=1548602)
     PhotoElectrons[1208](array_id=2, telescope_id=0)
     PhotoElectrons[1208](array_id=2, telescope_id=1)
     PhotoElectrons[1208](array_id=2, telescope_id=2)
     PhotoElectrons[1208](array_id=2, telescope_id=3)
     PhotoElectrons[1208](array_id=2, telescope_id=6)
     And 3 objects more of the same type
 MCPhotoelectronSum[2026](event_id=1548602)
 ArrayEvent[2010](event_id=1548602)
     TriggerInformation[2009](event_id=1548602)
     TelescopeEvent[2203](telescope_id=3)
         TelescopeEventHeader[2011](telescope_id=3)
         ADCSamples[2013](telescope_id=3)
         ImageParameters[2014](telescope_id=3)
         PixelList[2027](telescope_id=3, code=1, kind=selected)
     TelescopeEvent[2208](telescope_id=8)
         TelescopeEventHeader[2011](telescope_id=8)
         ADCSamples[2013](telescope_id=8)
         ImageParameters[2014](telescope_id=8)
         PixelList[2027](telescope_id=8, code=0, kind=triggered)
         PixelList[2027](telescope_id=8, code=1, kind=selected)
     TrackingPosition[2101](telescope_id=1)
     TrackingPosition[2102](telescope_id=2)
     TrackingPosition[2103](telescope_id=3)
     TrackingPosition[2104](telescope_id=4)
     TrackingPosition[2105](telescope_id=5)
     And 26 objects more of the same type
     StereoReconstruction[2015](result_bits=000000000101)
```

and the parsed trigger information:

```
In [5]: e["trigger_information"]
Out[5]: 
{'cpu_time': (1708711789, 292339000),
 'gps_time': (1708711789, 292339000),
 'trigger_pattern': 4,
 'data_pattern': 4,
 'n_triggered_telescopes': 1,
 'triggered_telescopes': array([3], dtype=int16),
 'trigger_times': array([10.260431], dtype=float32),
 'n_telescopes_with_data': 1,
 'telescopes_with_data': array([3], dtype=int16),
 'teltrg_type_mask': array([66], dtype=uint64),
 'teltrg_time_by_type': {},
 'plane_wavefront_compensation': {'az': 0.0,
  'alt': 1.2217304706573486,
  'speed_of_light': 29.972414016723633}}
```

So `telescopes_with_data` / `n_telescopes_with_data` doesn't match the reality, there are 2 telescope events (tel_ids 3 and 8):

Here are the two telescope event headers:

```

In [8]: e["telescope_events"][3]["header"]
Out[8]: 
{'loc_count': 443,
 'glob_count': 1548602,
 'cpu_time': (1708711789, 292339000),
 'gps_time': (0, 0),
 'trg_source': 1,
 'list_trgsect': array([332, 333, 337, 380, 384, 385]),
 'time_trgsect': array([41.259766, 41.503906, 41.503906, 41.748047, 41.503906, 41.748047],
       dtype=float32),
 'readout_time': 293.5289001464844,
 'relative_trigger_time': 9.033203125,
 'start_readout': 33,
 'telescope_id': 3}

In [9]: e["telescope_events"][8]["header"]
Out[9]: 
{'loc_count': 208,
 'glob_count': 1548602,
 'cpu_time': (1708711789, 292339000),
 'gps_time': (0, 0),
 'trg_source': 1,
 'list_trgsect': array([ 910,  911,  913,  920,  921,  922,  923, 1040, 1041]),
 'time_trgsect': array([56.25, 55.75, 56.5 , 56.  , 57.  , 55.5 , 55.75, 56.25, 56.75],
       dtype=float32),
 'readout_time': 168.71551513671875,
 'relative_trigger_time': 18.5,
 'start_readout': 37,
 'telescope_id': 8}

```

@kbernloehr @orelgueta Any idea what is going on here?

File is the same as in #2344: `/vo.cta.in2p3.fr/MC/PROD6/LaPalma/gamma/sim_telarray/4749/Data/000xxx/gamma_20deg_0deg_run000065___cta-prod6-2156m-LaPalma-dark+magic.simtel.zst`

maxnoe, 2024-11-28:
This happens three times in the file, everytime the stereo trigger information contains a single telescope, but then there are two telescope events:

```
2024-11-28 13:41:02,634 INFO [ctapipe.ctapipe-process] (tool.initialize): Loading config from '[]'
2024-11-28 13:41:02,635 INFO [ctapipe.ctapipe-process] (tool.initialize): ctapipe version 0.23.1.dev5+g05e5453d.d20241128
2024-11-28 13:41:02,636 INFO [ctapipe.ctapipe-process.SimTelEventSource] (eventsource.__init__): INPUT PATH = /home/maxnoe/Uni/CTA/data/grid-errors/gamma_20deg_0deg_run000065___cta-prod6-2156m-LaPalma-dark+magic.simtel.zst
2024-11-28 13:41:03,071 WARNING [ctapipe.ctapipe-process.DataWriter] (datawriter._setup_output_path): Overwriting /home/maxnoe/Uni/CTA/data/grid-errors/test.dl1.h5
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (process.start): applying calibration: True
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (process.start): (re)compute DL1: True
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (process.start): (re)compute DL2: False
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (process.start): compute muon parameters: False
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (subarray.info): Subarray : LaPalma-prod6
2024-11-28 13:41:03,212 INFO [ctapipe.ctapipe-process] (subarray.info): Num Tels : 32
2024-11-28 13:41:03,213 INFO [ctapipe.ctapipe-process] (subarray.info): Footprint: 0.65 km2
2024-11-28 13:41:03,213 INFO [ctapipe.ctapipe-process] (subarray.info): Height   : 2156.00 m
2024-11-28 13:41:03,214 INFO [ctapipe.ctapipe-process] (subarray.info): Lon/Lat  : -17.892030200000022 deg, 28.762166100000005 deg 
2024-11-28 13:41:03,214 INFO [ctapipe.ctapipe-process] (subarray.info): 
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info):        Type       Count Tel IDs
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info): ----------------- ----- -------
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info):    LST_LST_LSTcam     1 1      
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info): LST_MAGIC1_MAGIC1     1 31     
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info): LST_MAGIC2_MAGIC2     1 32     
2024-11-28 13:41:03,235 INFO [ctapipe.ctapipe-process] (subarray.info): MST_MST_NectarCam    26 5-30   
2024-11-28 13:41:56,892 WARNING [ctapipe.ctapipe-process.SimTelEventSource] (simteleventsource._generate_events): Encountered telescope event not present in stereo trigger information, skipping. event_id = 1548602, tel_id = 8, tels_with_trigger: [3]
2024-11-28 13:42:00,431 INFO [ctapipe.ctapipe-process.SimTelEventSource] (simteleventsource._generate_events): telescope event event_id = 1664106, tel_id = 32 is missing true image
2024-11-28 13:42:21,941 WARNING [ctapipe.ctapipe-process.SimTelEventSource] (simteleventsource._generate_events): Encountered telescope event not present in stereo trigger information, skipping. event_id = 2247909, tel_id = 14, tels_with_trigger: [32]
2024-11-28 13:43:17,227 WARNING [ctapipe.ctapipe-process.SimTelEventSource] (simteleventsource._generate_events): Encountered telescope event not present in stereo trigger information, skipping. event_id = 3974908, tel_id = 5, tels_with_trigger: [2]
2024-11-28 13:43:44,533 WARNING [ctapipe.ctapipe-process.SimTelEventSource] (simteleventsource._generate_events): Encountered telescope event not present in stereo trigger information, skipping. event_id = 4839806, tel_id = 22, tels_with_trigger: [1]
2024-11-28 13:43:48,932 INFO [ctapipe.ctapipe-process] (tool.run): Finished: ctapipe-process
2024-11-28 13:43:48,934 INFO [ctapipe.ctapipe-process] (tool.write_provenance): Output: 
2024-11-28 13:43:48,936 INFO [ctapipe.ctapipe-process.DataWriter] (datawriter.finish): Finishing output
```

maxnoe, 2024-11-28:
@kbernloehr Might this be the case where two telescopes trigger `RANDOM_MONO` in the same array event?

maxnoe, 2024-12-02:
I wrote a "frankenstein" file that includes the necessary parts to look into this / test behavior in ctapipe from the large original file and we added it to the data server:

http://cccta-dataserver.in2p3.fr/data/ctapipe-test-data/v1.1.0/prod6_issues.simtel.zst

kbernloehr, 2024-12-02:
Looking a the file with 'read_simtel -s' does confirm that there seems, at first glance, something wrong in event 1548602.
Only for telescope ID 3 there is a central trigger entry, as a random mono event with the misplaced bit pattern but both ID 3 (LST) and ID 8 (MST) have data. It seems that IDs 3 and 8 have, from the MC p.e. sums data block and the p.e. list data block, by far the most signal. With 265 and 338 true Cherenkov p.e. there might even be a chance for a genuine (rather than just a random mono) trigger. Given that the telescope event data reports 6 and 9 triggered sectors, both must have triggered by themselves. Were it not for the random mono flag, none of them would pass hardware stereo (ID 3) or software stereo (ID 8) trigger conditions. And no data would have been recorded for either of them. With the random mono activation of ID 3, the data for ID 8 are recorded as well, despite not having a valid contribution to the (stereo) array trigger. A somewhat similar situation may arise if there are two distinct groups of telescopes with non-overlapping software stereo, one of them passing the stereo trigger and the other only showing a single telescope. The data for that single telescope would be recorded as well.

In my preliminary analysis, the apparent discrepancy between reported array triggers - here only the random mono by ID 3 is valid - and the reported telescope triggers - for both ID 3, even before any random mono, and ID 8 - does not look like a bug to me. Perhaps not documented well enough, but it is the normal behaviour that, in the simulation at least, the single-triggered telescopes not participating in a successful array trigger are recorded as well. Whether their data would be available with the real instrument is open to interpretation but the MC stays on the safe side and makes it available. You can always neglect it afterwards but you cannot recover it if it is not recorded.

maxnoe, 2024-12-02:
@kbernloehr Thanks for checking.

So for now, would you say it's a good default behavior to discard the telescope event that is not in the central trigger object?

maxnoe, 2024-12-02:
> But a real array would have either an MST or an SCT in this location. No default choice here but it should be resolved after a sub-array selection.

This we have already covered with our software trigger. The error here turned out because we now want to select subarray only at a later stage to not have to redo lower-level operations per selected subarray. 
But then these "curious" events broke assumptions about the file structure / lengths of tables being the same between e.g. dl1 data and trigger tables.

Another option would be to include the additional telescope data and leave it to a later step to remove it (via `SoftwareTrigger`).

That would mean adding an entry to `event.trigger.tel` also for the telescope that doesn't appear in the stereo trigger.

> A somewhat similar situation may arise if there are two distinct groups of telescopes with non-overlapping software stereo, one of them passing the stereo trigger and the other only showing a single telescope. The data for that single telescope would be recorded as well.

Including that event in the reconstruction would be wrong, right? In real data we wouldn't have had this additional telescope (or we would have two subarray event ids for completely separate, valid triggers).

Is this possible in sim_telarray? Generating two separate stereo triggers for the same air shower? From the structure of the files / how event_ids are assigned I'd assume not.

> Not sure what Michael Punch would recommend here.

Let's ask. @mdpunch 

In short, I see three options:

1. Discarding the whole array event (I don't like this, should be done later using the SoftwareTrigger)
2. Discarding only the additional telescope event (Safe, but people might be interested in this data)
3. Keeping the additional telescope event, adding an entry to the tel trigger but not the stereo trigger info

3. requires a bit of thought... e.g. making sure that the `SoftwareTrigger` handles these events correctly and doesn't make a 2 telescope event from these that passes the criteria

maxnoe, 2024-12-03:
> So this is related to the RANDOM_MONO at somehow "promoting" the additional telescopes?

That's how I understood Konrad, yes.

> My immediate reaction would be to go with option 2.

That's also what I now implemented in #2659

closed 2024-12-04 by maxnoe