cta-observatory/ctapipe #2665: Implement Additional Muon Parameters in ctapipe for LST Analysis

jstvdk, 2024-12-03:
Currently, we are lacking some useful muon parameters that are used in processing LST-1 data and can be beneficial for subsequent analysis. I think it would be good to have them in `ctapipe` because we will need to use them in `calibpipe` for the derivation of calibration coefficients. For example, such muon metrics as:

- `ring_size` and `size_outside` - the sum of the photoelectrons in the ring area, defined as a region `dist_mask` = (R-0.25R; R+0.25R), and outside of it. 
- `mean_pixel_charge_around_ring` - average pixel charge in pixels surrounding the outer part of the ring
- `radial_stdev` - standard deviation of (cleaned) light distribution along ring radius
- 'radial_skewness' - skewness of (cleaned) light distribution along ring radius
- 'radial_excess_kurtosis' - excess kurtosis of (cleaned) light distribution along ring radius
- `num_pixels_in_ring` - number of pixels in the integration area `dist_mask`

Taking the above into account, I would like to:

- Add methods for calculating these parameters to the module `ctapipe.image.muon.features` (following already implemented solution in the `lstchain`).
- Add appropriate fields to the `MuonParameterContainers` to store the new parameters.
- Implement calculation of these parameters inside `MuonProcessor._calculate_muon_parameters()`

Please comment in case something is missing or we should follow other direction. Otherwise I can start to create corresponding PRs and implement these features.

maxnoe, 2024-12-03:
> ring_size and size_outside - the sum of the photoelectrons in the ring area, defined as a region dist_mask = (R-0.25R; R+0.25R), and outside of it.

We have something that is closely related: `intensity_ratio_inside_ring`:
https://github.com/cta-observatory/ctapipe/blob/11561812f28c8a7f9f77e4b08066cba4ccd3a32c/src/ctapipe/image/muon/features.py#L37-L44

But other than that, sounds good!

jstvdk, 2024-12-03:
> > ring_size and size_outside - the sum of the photoelectrons in the ring area, defined as a region dist_mask = (R-0.25R; R+0.25R), and outside of it.
> 
> We have something that is closely related: `intensity_ratio_inside_ring`:

Thanks for the comment, indeed this `intensity_ratio_inside_ring` looks good to isolate pure muon ring events apart from contaminated rings, which can currently partially be done by parameter `size_outside` in `lstchain`.
But apart from this, we still will need `ring_size` parameter for some of the `optical_efficiency studies`.

> But other than that, sounds good!

Cool, then I am starting to work on it.

PR #2670 (2025-03-27): Muon parameters update
I added new functionality to the muon analysis as described in the Issue #2665 , specifically

- new fields to the `MuonParametersContainer`
- new methods to the `ctapipe/image/muon/features.py`
- new tests for above mentioned methods in `ctapipe/image/muon/tests/test_muon_features.py`
- additional code block in `ctapipe/image/muon/processor.py` to process and save new muon parameters

Added changes mainly follow existing code in `lstchain/image/muon/muon_analysis.py`, with one exception:

- I redefined the variable `num_pixels_in_ring` - now it takes into account only p [... cut]
closed 2025-03-27 by kosack