cta-observatory/ctapipe #2685: Provenance config missing info for AstroQuantity values

kosack, 2025-01-21:
**Describe the bug**

The provenance log contains a `config` section with the results of `tool.get_curent_config()`.  However, many of the values e.g. for the `ctapipe-compute-irf` tool (see #2473 ) are `null` in the JSON output.  They are correct in the output of the function, so the problem is in conversion to JSON.  

```json
            "BackgroundRate2dMaker": {
               "fov_offset_max": null,
               "fov_offset_min": null,
               "fov_offset_n_bins": 1,
               "reco_energy_max": null,
               "reco_energy_min": null,
               "reco_energy_n_bins_per_decade": 5
            },
```

**To Reproduce**
Steps to reproduce the behavior:
1. run a tool that uses AstroQuantity config options
2. look at provenance.log and see missing values

**Expected behavior**

The JSON output contains the correct values, either as strings like `reco_energy_max="10 TeV"` or else as a structure like `reco_energy_max={"value":10, `unit`: "TeV"}`

**Supporting information**

Translations are done in the json dumper. So we need to add one for unit quantities.  Not sure why "null" is given, normally it should crash on unknown objects.

maxnoe, 2025-01-21:
The `Provenance.as_json` method defines a `set_default` method that returns `None` for unknown things, which will be serialized as null:

https://github.com/cta-observatory/ctapipe/blob/1644e85fe9449cbbf9b52ec28f6cb9fceee172f0/src/ctapipe/core/provenance.py#L371-L384

maxnoe, 2025-01-21:
I opened a draft PR here #2688, raising an error for unknown objects in `set_default`. That's probably not the behavior we want (a warning I think would be appropriate), but it nicely shows what currently is not supported:

https://github.com/cta-observatory/ctapipe/actions/runs/12889231768/job/35935972112?pr=2688

I can see the atmosphere enum and the astro quantity traits.

kosack, 2025-01-21:
Just noticed also that the `reference_metadata` in the `inputs` section is also `null`.  The ReferenceMetadata object can convert to JSON, but might be looking at an older version

maxnoe, 2025-08-25:
Fixed in #2688

closed 2025-08-25 by maxnoe