cta-observatory/ctapipe #2702: Error with ctapipe-optimize-event-selection when using the show config option

mdebony, 2025-02-19:
**Describe the bug**
When using the script ctapipe-optimize-event-selection with the --show-config option, the script do not produce the results file and it raise this error.

`2025-02-19 17:08:53,854 ERROR [ctapipe.ctapipe-optimize-event-selection] (tool.run): Caught unexpected exception: 'EventSelectionOptimizer' object has no attribute 'result'
Traceback (most recent call last):
  File "/home/mdebony/post_doc_CPPM/dpps/ctapipe/src/ctapipe/core/tool.py", line 440, in run
    self.finish()
  File "/home/mdebony/post_doc_CPPM/dpps/ctapipe/src/ctapipe/tools/optimize_event_selection.py", line 221, in finish
    self.result.write(self.output_path, self.overwrite)
    ^^^^^^^^^^^
AttributeError: 'EventSelectionOptimizer' object has no attribute 'result'
`
The same call without this option give the expected results.

maxnoe, 2025-02-19:
Where does this show-config option come from even? `ctapipe-process` doesn't have it..

maxnoe, 2025-02-19:
Aha. I assume it is because the process tool overrides `flags` and the optimize tool doesn't.

That's not great

kosack, 2025-03-17:
Also, If you want to see the config, look at the provenance log, which should have the correct one, including any defaults that were not specified explicitly.  The config returned by traitlets would be less useful (even if it worked...)

commit 0ca77cb8 (2025-02-19): Fix tool show-config, fixes #2702
closed 2025-03-24 by maxnoe