cta-observatory/ctapipe #2711: StereoMeanCombiner.predict_table() returns invalid table

TjarkMiener, 2025-03-12:
**Describe the bug**
The column "prefix_telescopes" is returned [here](https://github.com/cta-observatory/ctapipe/blob/main/src/ctapipe/reco/stereo_combination.py#L398-L405) with a list of tel ids which participated in the reco. However, it should be a boolean array of fix length. Code fails if you try to dump the returning table of `StereoMeanCombiner.predict_table()` via `write_table()`.

kosack, 2025-03-17:
The internal data model in ctapipe is to store telescope lists as real _lists_ of _tel_ids_.  The transformation to and from boolean arrays happens as a `ColumnTransform` in the `TableWriter`, which I guess is where we normally write these. `write_table` doesn't currently support general column transforms.   

Can you describe a bit more where this is happening, or what you are trying to do?

maxnoe, 2025-03-17:
But what @kosack writes is correct: the in-memory data structure expected is the list of tel ids. Only because tables currently does not support variable length arrays, we store it as a boolean mask.

kosack, 2025-03-17:
We could just do the same thing in write_table perhaps, similar to the way it's done for unit columns.