cta-observatory/ctapipe #2714: Segmentation fault when calling FlashCamExtrator

gschwefer, 2025-03-14:
I get segmentation faults when calling the `FlashCamExtractor `on simulated events. This had happened before, and appeared to be fixed by #2544, but now happened again. The error is not fully reproducible and comes and goes randomly, as was the case before #2544 . I have not fully located the problem, but suspect it must be the same numba accelerated function `adaptive_centroid` that caused the problem in #2544.

maxnoe, 2025-03-14:
> The error is not fully reproducible and comes and goes randomly, as was the case before https://github.com/cta-observatory/ctapipe/pull/2544

Not even on the same event?

gschwefer, 2025-03-14:
No, not even on the same event. And I've asked around, and others have reported this issue as well

kosack, 2025-03-17:
We run the FlashCamExtractor in the config we use to process all Prod5b/Prod6 files on DIRAC, right? Have we seen segfaults in any of those jobs?

maxnoe, 2025-03-17:
> We run the FlashCamExtractor in the config we use to process all Prod5b/Prod6 files on DIRAC, right? Have we seen segfaults in any of those jobs?

Prod5 we ran with 0.19.3 on the south with FlashCam. With the current version we didn't yet run with FlashCam as Prod6 South is not produced yet.

gschwefer, 2025-08-26:
I'm not surprised that it doesn't segfault for you..Like I said, even for me it only appears randomly, and even on the same event I can not reproduce it...
So could it be a problem with the computer (in this case the MPIK cluster) rather than a code issue in the end?

maxnoe, 2025-08-26:
Aside from the segfault issue: there are two lines in the code that don't do anything but look like they were intended to.

Inside of the while loops, `descend_limit` is assigned to conditionally, but to something that cannot have changed since the initial definition of `descent_limit` if I'm not wrong.

maxnoe, 2025-09-04:
I don't really expect this is directly related, as @gschwefer seems to have traced it to the `FlashCamExtractor`, which you don't use.

You stack shows it's failing to find a symbol in `scipy` and what makes me nervous is that you are apparently using a conda environment there is also a system library in the trace, so there could be a library version mismatch / ABI compatibility issue here.

Is there any LD_LIBRARY_PATH or similar set? Why does something inside of conda conda env suddenly reach out to a system library?

Also, are the packages all installed using conda? I don't see these object files in a conda-based installation where you see them, they seem to come only with PyPI wheels.

More information on the system architecture, operating system, installation method and any warnings you might have seen would be helpful.

jsitarek, 2025-09-04:
thx @maxnoe for quick feedback.
in CNAF I have magicctapipe installed in a conda environment (which also includes ctapipe and everything else needed)
On top of that environment I installed pystack with pip, and I see that it got installed inside the environment.
if I 'locate' libresolv it shows it in the system directory as in the log, but is it used anywhere in ctapipe? Maybe it is just something that pystack is using. I know that some libraries (I think this happens e.g. with some c++ libraries) cannot go into the conda environment. 
There is no LD_LIBRARY_PATH set in the environment 

architecture of the access machine:
cat /proc/version 
Linux version 5.14.0-503.14.1.el9_5.x86_64 (mockbuild@x64-builder01.almalinux.org) (gcc (GCC) 11.5.0 20240719 (Red Hat 11.5.0-2), GNU ld version 2.35.2-54.el9) #1 SMP PREEMPT_DYNAMIC Tue Nov 19 21:25:22 EST 2024

the installation was done by creating a conda environment from this file:
https://github.com/cta-observatory/magic-cta-pipe/blob/v0.6.0-alpha/environment.yml
and then installing MCP inside it:
https://github.com/cta-observatory/magic-cta-pipe/blob/v0.6.0-alpha/setup.cfg
I do not remember any errors from it (normally there are some packages reinstalled with different versions during this step). 
I actually tried it twice already to make sure that I did not make any error and both environments were giving problems, while making an environment based on ctapipe 0.19 did not produce crashes.

maxnoe, 2025-09-04:
@jsitarek 

I see numpy and scipy were installed by pip over the ones from conda and I suspect this is the issue

Elisa-Visentin, 2025-09-04:
I have it from pypy in the env I am using with slurm for mcs. But I am checking if I can fix this in the swtup file

maxnoe, 2025-09-04:
>  I will investigate at which step they get downloaded from pip.

I would assume that the conda env installs too recent versions of numpy and/or scipy and then the `pip install` of magic-ctapipe downgrades those packages.

But that leaves you I think with two versions of the shared libraries, the ones bundled with the pypi wheels and the ones in the conda environment and this _might_ create the issues you see.

Elisa-Visentin, 2025-09-04:
In conda we are not setting versions (well, for these two packages), so the conda version is probably inherited from lstchain and ctapipe. Now maybe we are managing to remove the 'pip overwriting'

maxnoe, 2025-09-04:
@jsitarek It down-graded numpy actually: from 2.2.6 to 1.26.4