cta-observatory/ctapipe #2718: Question: add JSONEncoder for TelescopePatternList?

mexanick, 2025-03-19:
**Please describe the use case that requires this feature.**

While trying to `json.dump` the configuration of the TelescopeComponent, the following error is produced:
```
TypeError: Object of type TelescopePatternList is not JSON serializable
```

**Describe the solution you'd like**

A custom encoder can be added somewhere to utils:

```python
class CustomJSONEncoder(json.JSONEncoder):
    """Encoder for TelescopePatternList."""

    def default(self, obj):
        """Convert TelescopePatternList to a list of tuples."""
        if isinstance(obj, TelescopePatternList):
            # Convert TelescopePatternList to a list of tuples
            return list(obj)
        return super().default(obj)
```

I'm not entirely sure where it would be best, and if it should be done in the repo, or next to the places where it is used.

maxnoe, 2025-03-19:
Yes, related to: #2688

maxnoe, 2025-08-25:
Done in #2688