cta-observatory/ctapipe #2721: Exlude metadata when joining tables

Hckjs, 2025-03-20:
This comes up in #2717 from some `tables.NaturalNameWarning` during `ctapipe-apply-models` when writing the tables with the `astropy_helpers.write_table` function: 
https://github.com/cta-observatory/ctapipe/blob/e5999e2667ddc7253dd86703a46b497fc3ae7ae0/src/ctapipe/tools/apply_models.py#L209-L214
and the "OBSGEO-..." metadata is added to the tables in the `TableLoader` with the `instrument=True` option:
https://github.com/cta-observatory/ctapipe/blob/e5999e2667ddc7253dd86703a46b497fc3ae7ae0/src/ctapipe/io/tableloader.py#L598-L601

maxnoe, 2025-03-20:
You could just do a `instrument_table.meta.clear()` before the join

maxnoe, 2025-03-20:
Alternatively, we could add a `meta_convention={"fits", "hdf5"}` option to the `to_table` methods instead of translating in `to_hdf5`

PR #2722 (2025-03-24): Adding `meta_convention` option to `SubarrayDescription.to_table()` method
As Discussed in #2717  and #2721.

closes #2721
closed 2025-03-24 by maxnoe