cta-observatory/ctapipe #2723: pyirf >=0.12.0 needs to be required, otherwise ctapipe-optimize-event-selection breaks

simonleeADL, 2025-03-21:
**Describe the bug**
When I installed the latest `ctapipe`, I had `pyirf 0.11.0` already cached. This however caused an error when running `ctapipe-optimize-event-selection`, as the `numpy` compatibility in `compat.py` didn't exist in `pyirf 0.11.0`, but it does in `0.12.0`.

Thus I think in `environment.yml` it should state `pyirf>=0.12.0`

**To Reproduce**
Steps to reproduce the behavior:
1. Have `pyirf 0.11.0` installed or cached
2. Install `ctapipe 0.23.2`
3. Run `ctapipe-optimize-event-selection`

maxnoe, 2025-03-21:
Hi @simonleeADL 

I am a bit confused, as you say `Install ctapipe 0.23.2`. The latest released version of ctapipe does not yet include anything using pyirf.

maxnoe, 2025-03-21:
And we do require pyirf with the version you mention in main:

https://github.com/cta-observatory/ctapipe/blob/e5999e2667ddc7253dd86703a46b497fc3ae7ae0/pyproject.toml#L58

simonleeADL, 2025-03-21:
Oops, my apologies, I mis-typed. Also I should've added more specific details.

I meant dev version, `0.23.3`, the one which has `ctapape-optimize-event-selection`.
And I was creating a conda environment with `mama create env -f environment.yml`. To my knowledge, conda doesn't yet support creating environments from a `pyproject.toml`.

closed 2025-04-30 by maxnoe