cta-observatory/ctapipe #2726: Spurious warnings about calibration_monitoring_id

kosack, 2025-03-24:
**Describe the bug**

For simualtions, there is no calibration_monitoring_id, but running ctapipe-process generates a large number of warnings about this column, which could be confusing to users.   This is because the `r1/event/telescope/tel_xxx` tables by default have this column, but it is set to None. 

```
2025-03-24 18:15:17,634 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column module_hires_local_clock_counter with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_009 not writable, skipping
2025-03-24 18:15:17,635 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column pedestal_intensity with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_009 not writable, skipping
2025-03-24 18:15:17,635 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column calibration_monitoring_id with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_009 not writable, skipping
2025-03-24 18:15:17,642 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column first_cell_id with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_014 not writable, skipping
2025-03-24 18:15:17,642 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column module_hires_local_clock_counter with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_014 not writable, skipping
2025-03-24 18:15:17,642 WARNING [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._create_hdf5_table_schema): Column pedestal_intensity with value None of type <class 'NoneType'>  of container R1CameraContainer in table r1/event/telescope/tel_014 not writable, skipping
...
```

Maybe should set the default value to some nominal value? or just mark that column to be skipped if this is a simulation.

**To Reproduce**
Steps to reproduce the behavior:

```sh
ctapipe-process -i dataset://gamma_prod5.simtel.zst -o event.h5  --DataWriter.write_r1_waveforms=True --progress
```

**Expected behavior**

No warnings for simulations.

maxnoe, 2025-03-24:
Maybe we could just actually store the applied r1 calibration and assign an id to that?

Probably the run id makes most sense

kosack, 2025-03-28:
> Probably the run id makes most sense

Yeah, for simulations maybe that is sufficient. For real data, this will be filled in, so shouldn't be a problem, and the ID will come from ACADA's database or something like that.