cta-observatory/ctapipe #2730: DataWriter: Caught unexpected exception: 'float' object has no attribute 'to_value'

burmist-git, 2025-04-02:
**Describe the bug**
Looks like we can not write the muonparameters_skewness during the muon analysis 
Error writing col muonparameters_skewness of container MuonParametersContainer

**To Reproduce**
ctapipe-process --overwrite --input=$in_simtel_file --output=$out_r0_dl1_h5_file --config=$config_file

$config_file:
DataWriter:
  output_path: None
  write_muon_parameters: False
  write_dl1_parameters: False
  write_dl1_images: True
  overwrite: True

MuonProcessor:
  completeness_threshold: 30
  ratio_width: 1.5
  pedestal: 1.1

MuonIntensityFitter:
  hole_radius_m: [["type", "LST_*", 0.308]]
  spe_width: 0.5
  min_lambda_m: 0.0000003
  max_lambda_m: 0.0000006
  oversampling: 3

ImageParameterQuery:
  quality_criteria: [
      ["min_pixels", "dl1_params.morphology.n_pixels > 100"],
      ["min_intensity", "dl1_params.hillas.intensity > 500"]
    ]

RingQuery:
  quality_criteria: [
      ["radius_not_nan", "np.isfinite(ring.radius.value)"],
      ["min_pixels", "np.count_nonzero(mask) > 50"],
      ["ring_containment", "parameters.containment > 0.5"]
    ]

ImageProcessor:
  image_cleaner_type: "TailcutsImageCleaner"
  use_telescope_frame: True
  apply_image_modifier: False

TailcutsImageCleaner:
  picture_threshold_pe : 10
  boundary_threshold_pe : 5
  min_picture_neighbors :  2
  keep_isolated_pixels : False

TimeConstrainedImageCleaner:
  time_limit_core_ns: 4.5
  time_limit_boundary_ns: 1.5

ImageQualityQuery:
  quality_criteria: [["size_greater_0", "image.sum() > 0"]]

CameraCalibrator:
  data_volume_reducer_type: "NullDataVolumeReducer"
  image_extractor_type: "GlobalPeakWindowSum"
  invalid_pixel_handler_type: "NeighborAverage"
  apply_waveform_time_shift: False
  apply_peak_time_shift: True

GlobalPeakWindowSum:
  window_width: 7
  window_shift: 3
  apply_integration_correction: True
  pixel_fraction: 1

**Expected behavior**
normal operation with output file together with muon parameters

burmist-git, 2025-04-02:
@mexanick ;-) You asked to look on the  issue - this is simple reminder ...

maxnoe, 2025-04-02:
@burmist-git could you please provide the full error message?

burmist-git, 2025-04-02:
@maxnoe sure !!

2025-04-02 17:37:45,060 WARNING [ctapipe.ctapipe-process.DataWriter] (datawriter._setup_output_path): Overwriting /home/burmist/home2/work/CTA/cta-lstchain/cta-lstchain_mu_images_ana/data/run_000/dl1_muon_ctapipe_run000.h5
/home/burmist/miniconda/envs/mamba/envs/cta-dev/lib/python3.11/site-packages/astropy/units/quantity.py:671: RuntimeWarning: invalid value encountered in sqrt
  result = super().__array_ufunc__(function, method, *arrays, **kwargs)
2025-04-02 17:37:47,166 ERROR [ctapipe.ctapipe-process.DataWriter.HDF5TableWriter] (hdf5tableio._append_row): Error writing col muonparameters_skewness of container MuonParametersContainer
2025-04-02 17:37:47,166 ERROR [ctapipe.ctapipe-process] (tool.run): Caught unexpected exception: 'float' object has no attribute 'to_value'
Traceback (most recent call last):
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/core/tool.py", line 445, in run
    self.start()
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/tools/process.py", line 315, in start
    self.write(event)
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/datawriter.py", line 340, in __call__
    self._write_muon_telescope_events(event)
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/datawriter.py", line 677, in _write_muon_telescope_events
    self._writer.write(
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/hdf5tableio.py", line 491, in write
    self._append_row(table_name, containers)
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/hdf5tableio.py", line 457, in _append_row
    value = self._apply_col_transform(table_name, colname, value)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/tableio.py", line 199, in _apply_col_transform
    value = tr(value)
            ^^^^^^^^^
  File "/home/burmist/home2/work/CTA/ctapipe/src/ctapipe/io/tableio.py", line 349, in __call__
    return value.to_value(self.unit)
           ^^^^^^^^^^^^^^
AttributeError: 'float' object has no attribute 'to_value'

PR #2732 (2025-04-05): Force absences of units for skewness and excess_kurtosis
Explicitly convert (dimensionless) skewness and excess_kurtosis to values

Closes #2730
closed 2025-04-05 by kosack