cta-observatory/ctapipe #2737: Difficult with disp reconstruction for mono SST

simonleeADL, 2025-04-11:
Hi there!

I'm attempting to create IRFs of a single SST using the `ctapipe 0.24.0` CLI.

After disp reconstruction, many point-source gamma-rays are well-reconstructed, but many are not, clustered in a ring around the true source direction. The resulting angular resolution is thus very poor.

I don't know how best to diagnose this. My first thoughts are that it maybe has something to do with sign reconstruction, or perhaps coordinate transformations in a dual-mirror telescope?

[image]

[image]

Here is my `train_disp_reconstructor.yaml`

```
TrainDispReconstructor:
  random_seed: 0
  n_events:

  CrossValidator:
    n_cross_validations: 5
    rng_seed: 1337

  DispReconstructor:
    n_jobs: -1

    norm_cls: RandomForestRegressor
    log_target: True
    sign_cls: RandomForestClassifier

    QualityQuery:
      quality_criteria:
        - ["enough intensity", "hillas_intensity > 50"]
        - ["Positive width", "hillas_width > 0"]
        - ["enough pixels", "morphology_n_pixels > 3"]
        - ["not clipped", "leakage_intensity_width_2 < 0.5"]

    FeatureGenerator:
      features:
        - ["area", "hillas_width * hillas_length"]

    features:
      - hillas_intensity
      - hillas_r
      - hillas_length
      - hillas_width
      - hillas_skewness
      - hillas_kurtosis
      - timing_slope
      - timing_deviation
      - leakage_intensity_width_1
      - leakage_intensity_width_2
      - leakage_pixels_width_1
      - leakage_pixels_width_2
      - concentration_cog
      - concentration_core
      - morphology_n_islands
      - morphology_n_large_islands
      - morphology_n_medium_islands
      - peak_time_std
      - peak_time_skewness
      - peak_time_kurtosis
      - intensity_max
      - intensity_mean
      - intensity_std
      - intensity_skewness
      - intensity_kurtosis
      - area
```

maxnoe, 2025-04-11:
Hi @simonleeADL,

> After disp reconstruction, many point-source gamma-rays are well-reconstructed, but many are not, clustered in a ring around the true source direction. The resulting angular resolution is thus very poor.
> 
>  My first thoughts are that it maybe has something to do with sign reconstruction,

Correct.

Yes, this is – in general – expected for disp reconstruction with monoscopic telescopes. The ring you see are the events with wrongly reconstructed disp sign, often referred to as "ghosts" (thus the "ghost busting" term for the disp-sign reconstruction also known as head-tail disambiguation.

I don't know enough about SST analysis to have an intuition about the expected performance, so I cannot really say if this is much worse than it could be and thus hint of a bug or issue with dual mirror telescopes in ctapipe. The general effect will always be there for monoscopic direction reconstruction using a disp estimator.

An observation is that the numbers you present as angular resolution do not seem to fit the image and indeed are much worse than I would expect.

In general, especially for low energy events, disp sign prediction will always be bad, at the threshold close to randomly guessing.
You can make a couple of sanity checks. E.g. plot the accuracy of disp sign prediction vs. true energy or vs. hillas intensity.
Check the feature importances of your model, the most important features for sign should be time slope and hillas skewness.

Since this effect to some degree will always affect the angular resolution, a common convention is to not include the wrongly classified disp sign events when computing effective area, instead treating them as "lost". This effectively moves the issue to the effective area. So instead of a bad angular resoltuion that doesn't really describe the real situation around the source location, you get a lower effective area.

See for example the LST performance paper:

> The gamma-ray point-spread function for a single IACT has a central component
made of events with properly determined head-tail image orientation (i.e. correctly reconstructed disp_sign, see Figs.
3 and 4), and, especially at low energies, a separate tail made up by events with wrong orientation. Near threshold
the fraction of correctly oriented images is less than 60%, but it increases fast with energy (see section 3.6)

https://arxiv.org/pdf/2306.12960

simonleeADL, 2025-04-11:
Hi @maxnoe !

Thankou for the response.

I'm away from my computer at the moment,  so apologies for the lack of further plots.

I think the az/alt plot is somewhat visually deceiving. The percentage of events reconstructed away from the true point is very large, increasing towards 100% as energies get higher (which, as you've said, is the opposite of what is expected).

In my own analysis I've found a mono SST to achieve ~0.2 deg angular resolution.

With the timing information available, I've typically found sign reconstruction performance to be near-perfect for high energies.

It's entirely possible that I've misconfigured something about the ML training or application, but I've tried to keep it fairly in-keeping with the `ctapipe-quickstart` defaults.

Whereabouts in the ctapipe code would I look for any potentially relevant coordinate transformations or sign classification during the direction reconstruction model application?

I'll do the suggested sanity checks once I'm back in the office!

Again, thankyou for the response!

maxnoe, 2025-04-11:
> I think the az/alt plot is somewhat visually deceiving. The percentage of events reconstructed away from the true point is very large, increasing towards 100% as energies get higher (which, as you've said, is the opposite of what is expected).

That's certainly a sign of a bug! But what surprises me a bit is that this is a machine learning algorithm... if there is a bug in the coordinate transformation affecting disp sign, shouldn't it be symmetric in training and application and thus cancel out?

> It's entirely possible that I've misconfigured something about the ML training or application, but I've tried to keep it fairly in-keeping with the ctapipe-quickstart defaults.

I doubt that's the case! The performance is so bad for what your descibre that it cannot be explained by a somewhat suboptimal ML configuration.

> Whereabouts in the ctapipe code would I look for any potentially relevant coordinate transformations or sign classification during the direction reconstruction model application?

So the general transformations between `AltAz` and `TelescopeFrame` live here:
but there should not be anything specific to dual mirror telescopes here, as that's still "field of view" coordinates.

Even the transformation from `CameraFrame` to `TelescopeFrame` has nothing specific to dual mirror telescopes, as that is defined like that. Only in the `EngineeringCameraFrame` there is a difference between Dual mirror and single mirror telescopes.

The code concerned with disp estimation lives here:
https://github.com/cta-observatory/ctapipe/blob/ba264d9685e58b748c65d1d89c66ae0365588103/src/ctapipe/reco/sklearn.py#L824-L850

Again, I wouldn't assume there to be something specific for dual mirror telescopes here...

Is what you show here done purely on simulations? How do you execute the tools? What data do you use for training / application etc.?

maxnoe, 2025-04-11:
Ah, but I see you are not passing any options to the sklearn class, like `n_estimators` or `max_depth`. But recent sklearn has quite reasonable defaults for the random forest (it used to be just 10 trees, but now it's 100)

moralejo, 2025-04-11:
> 
> I think the az/alt plot is somewhat visually deceiving. The percentage of events reconstructed away from the true point is very large, increasing towards 100% as energies get higher (which, as you've said, is the opposite of what is expected).
> 

Funny... But then, what is the population of very well reconstructed events in your Alt-Az plot, if not high energies? I would start by checking how exactly the angular resolution plot is made, from the Alt-Az plot it does not seem like the problem is the RF or in the coordinate transformation.

simonleeADL, 2025-04-11:
@moralejo 
The angular resolution plot in the 68th percentile column as created with `ctapipe-compute-irf`. But also I'm confident it would be very similar if I just got the data after `ctapipe-apply-models`, calculated the angular separation for each event, and plotted the 68th percentile (given the state of the alt/az distributions).

From memory I think the largest proportion of well-constructed events were around ~1 TeV (above the energy threshold, and before the reconstruction got worse at higher energies) but they still did not make up the majority.

@maxnoe 
Everything is CORSIKA/sim-telarray simulations that I've run. The training data is diffuse gamma rays, the test data is point-source with a varity of pointing offsets (the plots before are for 1 deg offset in the sky, but I was getting the same for no offset). The tools were all the ctapipe CLI tools – `ctapipe-process`, `ctapip-train-disp-reconstructor`, `ctapipe-apply-models` (and then also `ctapipe-optimize-cuts` and `ctapipe-compute-irf`, but that obviously doesn't change the disp results) using as close to default configs as I could (aside from removing stereo paramenters and changing to RF). And yes, I also noticed that by default sklearn's RF uses 100 trees, which is nice!

Also, I should say, energy reconstruction seemed to work as expected! The confusion matrix looked good.

I'm basically just trying to get a fairly bog standard analysis pipeline from simtel to irf working.

I guess also without a colourbar that alt/az plots doesn't properly get across that the whole space is kinda bathed in poorly reconstructed event, apologies!

maxnoe, 2025-04-11:
@simonleeADL Could you somehow share your DL1 / DL2 files and the configuration so we can have a look?

maxnoe, 2025-04-11:
> Does this work?

Yes, thanks. Though I see this is tiny... is this all the events you have for training / evaluation?

simonleeADL, 2025-04-11:
@maxnoe 
I think there's ~40,000 diffuse events (before quality cuts, don't remember after) to train on, and ~95,000 (before quality cuts) point source gammas to test on. I think that should sufficient for a working test?

maxnoe, 2025-04-11:
Sounds good, yes

moralejo, 2025-04-11:
> [@moralejo](https://github.com/moralejo) The angular resolution plot in the 68th percentile column as created with `ctapipe-compute-irf`. But also I'm confident it would be very similar if I just [...]

Is the same code tested on other samples / telescope types?  If not, might it be that the disp-sign is just used wrongly ("inverted") in the _application_ of the RF?

maxnoe, 2025-04-11:
From running the train step:
```
2025-04-11 16:06:21,972 INFO [ctapipe.ctapipe-train-disp-reconstructor.CrossValidator] (sklearn.__call__): Mean R^2 score from CV: 0.9215 ± 0.0057
2025-04-11 16:06:21,972 INFO [ctapipe.ctapipe-train-disp-reconstructor.CrossValidator] (sklearn.__call__): Mean accuracy score from CV: 0.0291 ± 0.0019
```

This looks like a sign-flip in the application.

I'll investigate a bit more and compare with LST/MST sims

maxnoe, 2025-04-11:
Seems this is just broken, I get sub-50% accuracy also for LST and MST.

cc @LukasBeiske

maxnoe, 2025-04-11:
The disp sign was broken when we introduced the disp scoring feature in this PR:
https://github.com/cta-observatory/ctapipe/pull/2479

There is no test for the actual performance of the prediction and it is currently not used in our standard grid analysis, so it went unnoticed.

closed 2025-04-14 by maxnoe

simonleeADL, 2025-04-15:
@maxnoe Fantastic! All is looking good now. Thankyou!

[image]

[image]

maxnoe, 2025-04-16:
@simonleeADL 0.25 has been released with the fix included

PR #2738 (2025-04-14): Fix wrong proba column being used for disp sign
fixes #2737