cta-observatory/ctapipe #2761: Empty DL2 Tel Containers from HDF5EventSource

Hckjs, 2025-05-27:
Reading in data from the `DL2_TELESCOPE_GROUP` with the `HDF5EventSource` leads to default filled/empty dl2 telescope containers for DL2 files written with the `ApplyModels` Tool. 
I think the `HDF5TableReader.read()`-Method is missing a `prefixes=(algorithm + "_tel",)` argument in the `HDF5EventSource` generator for the DL2_TELESCOPE_GROUP here:
https://github.com/cta-observatory/ctapipe/blob/e6eeae9747e96d4e021f661da36f80f4c5027694/src/ctapipe/io/hdf5eventsource.py#L600-L606

just like its written in the DL2_SUBARRAY_GROUP here:
https://github.com/cta-observatory/ctapipe/blob/e6eeae9747e96d4e021f661da36f80f4c5027694/src/ctapipe/io/hdf5eventsource.py#L578-L585

because for prefix=None the TableReader tries to map the containers field_names to the tables column_names via its column_attribute "NAME" which doesn't exist for the DL2 Tables not written with the DataWriter. So we can either adapt 
https://github.com/cta-observatory/ctapipe/blob/e6eeae9747e96d4e021f661da36f80f4c5027694/src/ctapipe/reco/utils.py#L1

to add the missing NAME attribute or add the above mentioned prefix.

maxnoe, 2025-08-26:
Ah, might just be a missing `prefixes=` argument in the HDF5EventSource.

maxnoe, 2025-08-27:
The analysis is mostly correct, the column metadata needed to read these fields back into containers was not added.

But what I wrote above is also correct: we can also explicitly add the prefixes when reading.

In #2824 I added both fixes actually, so that data we write now has the full metadata, but we also have the workaround to fix reading existing files.

PR #2824 (2025-09-04): Implement reading and writing column metadata for tables. Fix reading of dl2 ml tel output in HDF5EventSource
Fixes #2761
closed 2025-09-04 by kosack