cta-observatory/ctapipe #2768: Support for random mono trigger in ctapipe/pyeventio

mdpunch, 2025-06-06:
I have been using `len(event.trigger.tels_with_trigger)==1` as a sign that I'm looking at a `RANDOM_MONO_PROB` event in DL1 files (and earlier in the chain also).  

This is fine when looking at the full array simulations, even though they include different telescope test positions, NectarCAMs on top of FlashCams, MAGIC telescopes, etc.

But when looking at a subarray (e.g. the alpha configuration), this doesn't work, because the event could be a stereo trigger (for simtelarray) with a telescope which isn't in the subarray.

For some studies I would like to "recreate" what the full response (mono+stereo) of a telescope in a subarray, which I did by taking the supposed mono events together with 1/RANDOM_MONO_PROB of the stereo events.  But this will be incorrect for subarrays, because the supposed mono events are actually stereo.

I can get around this by opening the file twice, once with the required subarray, and again with the full simulated array, and stepping through the events so that with the latter I can choose "real" random mono events.  Or possibly by opening the file with the full array and grabbing the mono event IDs for later use

It would be more practical to have just an event type for these random mono events, as discussed here: https://gitlab.cta-observatory.org/cta-science/simulations/request-new-simulations/-/issues/12#note_105670

As @maxnoe mentioned in the closing comment on  https://github.com/cta-observatory/pyeventio/issues/261,
> So that's something to be added.

Is this something I should work on, or a job for ctapipe/pyeventio people?

maxnoe, 2025-06-06:
@mdpunch Could you confirm that what we want is:

* Add a specific event type for the telescope events only triggered via random mono
* Remove these events from the subarray event in our `SoftwareTrigger`.

maxnoe, 2025-06-06:
From io_hess.h:

```
   int teltrg_type_mask[H_MAX_TEL]; ///< Bit mask which type of trigger fired.
                        ///< More than one trigger type per telescope is possible.
                        ///< Bits well beyond H_MAX_TRG_TYPES are modifier flags, in addition to trigger type(s).
                        ///< Bit 0: (mostly analog) majority trigger.
                        ///< Bit 1: analog sum trigger.
                        ///< Bit 2: digital sum trigger (working on readout data stream).
                        ///< Bit 3: digital (majority) trigger (working on separately digitized data).
                        ///< Bits 4-7: reserved (must be zero).
                        ///< Bit 8: long-event modifier (readout may include more samples than normal).
                        ///< Bit 9: matching muon-ring enhancement conditions, acceptable mono event.
                        ///< Bit 10: randomly chosen as acceptable mono event.
                        ///< Bits 11-15: unspecified (not used so far).
                        ///< Bits 16-31: reserved (must be zero).
```

This doesn't seem to fit what I see in the file, seems to be shifted by one hex order of magnitude (0x41 instead of 0x401).

maxnoe, 2025-06-06:
What is the correct thing to do for subarray events that contain random mono prob triggered telescope events?

E.g. from an example file I simulated with relatively high random mono prob, this is a triggered event: 

```
tel_ids:                [ 10  14     15  24]
trigger patterns: [   1    1 1025    1]
```

mdpunch, 2025-06-07:
> What is the correct thing to do for subarray events that contain random mono prob triggered telescope events?
>
> E.g. from an example file I simulated with relatively high random mono prob, this is a [...]

Thanks for this work, Max @maxnoe !

So, to see if I understand correctly Konrad @kbernloehr adds a bit to the `teltrg_type_mask` for `simtel`, which is good.  But there was a bug, which is fixed now, and ctapipe can detect which files were produced with the bug, and correct for it, luckily.

For your last question, I didn't consider that this was possible, to get a random mono telescope together with an array event! (Though in fact this can happen in real life, as I had forgotten I had shown in 2019, until Adam reminded me https://indico.cta-observatory.org/event/2249/contributions/21603/attachments/16315/21323/TATS_201910_CTAC.pdf , page 18).  But that's not what we want in this context!

If a telescope has triggered along with some other telescopes in the event, how is that a `random_mono`, rather than just one telescope in a stereo event?  Or does the random_mono bit being set just mean "this event would have been chosen as a `random_mono` trigger, except it happens to be part of a stereo array event"?

I and some other people would like to use these as "this is a `random_mono` event which wasn't part of a stereo event, so a *mono-only* event, so you can combine it with `1/random_mono_prob` of the stereo events for this telescope (or that fraction of their distribution) to get the proper stats (or distribution) for *full-mono* for this telescope".

So, the way to treat this depends a bit on what exactly has been implemented.  But, with a `random_mono_prob==0.01` as is being used in prod6, the effects are marginal...

mdpunch, 2025-06-07:
I see from the comments in the code in the proposed #2769:
```
    #: Random mono trigger, used in simulations to force storage
    #: of event regardless of subarray trigger for certain studies.
    RANDOM_MONO = 18
```
... that the implementation **must** be as stated (i.e. "regardless of subarray trigger"), but better if Konrad @kbernloehr could confirm this.

But this is fine, also, as long as we know...  In this case, for my treatment above I would check that `RANDOM_MONO = 18` and also `len(telsecopes_triggered)==1` to ensure my *mono-only* sample, no problem.  Indeed, I could do this whatever the interpretation of `random_mono`, to be sure.

mdpunch, 2025-06-11:
Some conclusions after discussion with @maxnoe: 

- The "promotion" of a telescope to a _random_mono_ trigger (even if part of a stereo event) is fine as a way to work
  * Indeed, actual telescopes could do this if needed, whereas having _random_mono_but_not_stereo_ isn't really physical 
- When choosing a subarray, for normal stereo operation the user should always call the `SoftwareTrigger` class, as Max says "we have an additional class in ctapipe to remove these events (and also simulate the LST Hardware trigger condition of at least two LSTs):
https://ctapipe.readthedocs.io/en/latest/api/ctapipe.instrument.SoftwareTrigger.html#ctapipe.instrument.SoftwareTrigger "
- In order to choose _random_mono_only_ events in a subarray this can be done with the condition `random_mono and len(tels_triggered)==1`
  * The aim for such studies being to combine _random_mono_only_ events from a telescope with the higher-statistics _stereo_events_ from that telescope (either by scaling the stereo distribution or randomly choosing `1/random_mono_probability` of the stereo events, according to the need).

So, I think I can close this, and communicate the conclusions to the interested parties doing this kind of study.

closed 2025-06-11 by mdpunch