cta-observatory/ctapipe #2786: DispReconstructor is lacking check on the pointing frame at inference time

mdebony, 2025-06-24:
**Describe the bug**
The DispReconstructor is lacking a check at inference time on the pointing frame if pointing information are taken from the observation block. Therefore it will perform the reconstruction like it is Alt Az coordinates even if it's ICRS or Galactic coordinates.
https://github.com/cta-observatory/ctapipe/blob/4e55c3e041790be0a1b5a14f814ebbb136e1d306/src/ctapipe/reco/sklearn.py#L834

**To Reproduce**
1. Apply DispReconstructor on data with the pointing frame set as ICRS or Galactic

**Expected behavior**
Two options (to be discussed) :
- Raise an exception
- Convert ICRS and Galactic to AltAz

mdebony, 2025-06-24:
@maxnoe @kosack  I will make the fix, but what is the wanted behavior. Converting the coordinates to AltAz will be more versatile but raising an exception will be simpler.

maxnoe, 2025-06-24:
@mdebony I opened #2785 for now.

Doing the conversion is relatively simple if we use the subarray reference location, but I think at least for patanal this would lead to signing errors.

Using the telescope positions is possible but a bit more involved.

kosack, 2025-06-25:
Maybe for now at least just raise the exception. In general, how we treat pointing and coordinates needs a bit of an overhaul, not just in the DispReconstructor.  Perhaps we can open a more general issue or use the _Discussions_ board to discuss pointing in general?

mdebony, 2025-06-25:
Ok, @kosack I let you open it if it's ok with you as you are bit more aware than me on the possible issues link to that point.

maxnoe, 2025-08-25:
Fixed in #2785

closed 2025-08-25 by maxnoe