cta-observatory/ctapipe #2794: ctapipe-quickstart prints many warnings

Tobychev, 2025-07-02:
**Describe the bug**
`ctapipe-quickstart` prints many warnings 

**To Reproduce**
Run `ctapipe-quickstart`.

**Expected behavior**
That no warnings about ctapipe internals should arise.

**Supporting information**
```
❯ ctapipe-quickstart
Enter your contact name: Tomas Bylund
Enter your contact email: tomas.bylund@obspm.fr
Enter your organization: Observatoire de Paris - Meudon, APC

/Users/tbylund/miniforge3/envs/ctapipe/lib/python3.12/site-packages/ctapipe/core/provenance.py:280: MissingReferenceMetadata: Could not read reference metadata for input file: /Users/tbylund/develop/dpps/Work/base_config.yaml
  warnings.warn(
/Users/tbylund/miniforge3/envs/ctapipe/lib/python3.12/site-packages/ctapipe/core/provenance.py:280: MissingReferenceMetadata: Could not read reference metadata for input file: /Users/tbylund/develop/dpps/Work/calculate_pixel_stats.yaml
  warnings.warn(
/Users/tbylund/miniforge3/envs/ctapipe/lib/python3.12/site-packages/ctapipe/core/provenance.py:280: MissingReferenceMetadata: Could not read reference metadata for input file: /Users/tbylund/develop/dpps/Work/stage1_config.yaml
[snip]
```

**Additional context**
```
❯ ctapipe-info --version

*** ctapipe version info ***

version: 0.26.0
```

kosack, 2025-07-03:
Ah, yes that is just because the provenance `add_input()` now tries to get reference metadata, but of course here the files are just text files with no metadata. I'll chance it to not try to get ref metadata for those.

Tobychev, 2025-07-03:
Sure, or else the quickstart, who is the one making these filese, could maybe provide the metadata itself somehow?

commit 1d18e327 (2025-07-22): Fix warnings about missing meta in quickstart tool, fixes #2794
commit 2b5313c4 (2025-07-22): Merge pull request #2797 from cta-observatory/fix_quickstart_warnings

Fix warnings about missing meta in quickstart tool, fixes #2794
closed 2025-07-22 by maxnoe