cta-observatory/ctapipe #2810: Energy resolution/dispersion  with irf maker tool

BastienLacave, 2025-08-15:
Hi,

I am using ctapipe irf maker tool on CTLearn data, that is in the ctapipe format. First I optimize the cuts for 70% gamma efficiency, then compute IRFs (see config), and no errors/warnings are shown.
I have a sample of LST-1 MC gammas from which I compute IRFs.
The effective area and angular resolution look good, but the energy resolution is all over the place... (see plots below).

To check, I took the optimized cuts from the cuts file and cut the data myself, then used pyirf and ctaplot to compute the energy resolution, and it looks good. I tested 90% and 40% gamma efficiencies as well and the results are similar.

I then suspect the the irf tool is making something wierd for the energy resolution. Do you know what could cause that ?
Thanks a lot for your help,
Bastien

[image]

[image]

[public-conf_eff70.yml](https://github.com/user-attachments/files/21790703/public-conf_eff70.yml)

maxnoe, 2025-08-15:
Is that combining all offsets together? the ctapipe tool computes in bins of FoV offset.

What offset bin are you showing?

maxnoe, 2025-08-15:
> LST-1 MC gammas 

Are the observation ids / event ids unique? LST MC productions have issues here.

maxnoe, 2025-08-26:
Could you post plots of the two energy dispersion matrices? Preferably with an energy axis not just a bin index?

Looking by eye, it seems like the energy resolution matches the dispersion matrix, so the problem might not actually be in the resolution computation.

But what would really help is if you could provide your data files.

maxnoe, 2025-08-26:
Please provide the exact commands you are running and the config files.

Naively running `ctapipe-optimize-event-selection` and then `ctapipe-compute-irfs`, I get errors and cannot even look at an energy resolution.